jeremiah-c-leary/vhdl-style-guide #815: Setting style  to "no_blank_line"

njm295, 2022-08-08:
Setting the global Rule "style" set to "no_blank_line" changes the formatting ( white space)
of constant, type and signal declarations.

[image]

jeremiah-c-leary, 2022-08-09:
Hey @njm295,

That would be the expected behavior when setting `style` to "no_blank_line" in `global`.  The rule `type_010` handles blank lines above the `type` keyword and rule `type_011` handles blank lines below the `type` keyword.  

So in your case VSG is removing the blank lines.  If you would like to preserve the blank lines before and after the `type`, then add the following to your configuration:

```json
"rule":{
  "type_010": {
     "style":  "require_blank_line"
  },
  "type_011": {
     "style": "require_blank_line"
  }
}

Regards,

--Jeremy

njm295, 2022-08-09:
@jeremiah-c-leary 
The issue is not about the blank lines but about the white space between the (constant, signal, type , ...) declarations  and their identifiers.   I had noticed that after  fixing  my file with the intended correction  of the blank line rule; the spaces between these declarations and their identifiers changed to  align all the identifiers -- I do not want that.   I want the identifier alignment to be shared across all declaration types.  I want  identifiers to be aligned uniquely with in each declaration type..
Thanks,
nick

jeremiah-c-leary, 2022-08-09:
Hey @njm295,

Ah...I misunderstood.  Thanks for the clarification.

There are several rules which attempt to align keywords into columns.  In this particular case you should look at the following rules:

* architecture_027
* architecture_029
* architecture_400

`architecture_029` aligns identifiers and `architecture_027` aligns colons.  If you disable those two rules:
```json
{
  "rule" : {
     "architecture_027" : {
        "disable" : "True"
     },
     "architecture_029" : {
      "disable" : "True"
     }
  }
}
```

That will stop the alignment.

There are additional rules which perform this type of alignment and you can find a list of them in the [documentation](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_keyword_alignment_rules.html#rules-enforcing-keyword-alignment).

--Jeremy

njm295, 2022-08-09:
@jeremiah-c-leary 
Thanks.  still having difficulty getting the identifiers to be aligned uniquely with in each type.

I stumbled on to this when changing the style to no_blank_line changed the  default keyword aliugnment.  I attach a figure or what I currently have and would like. (please see lower right panel)

[image]

jeremiah-c-leary, 2022-08-12:
Hey @njm295,

I see now...I will have to think a little bit on the best way to implement that option.  I have two options and I would want to ensure the option I choose fits with the other rules.

--Jeremy

jeremiah-c-leary, 2022-10-09:
Evening @njm295 ,

I apologize that it has taken me this long to get back to this issue.

I added some rules that I believe with give you the formatting you are looking for.  I took your original code for the architecture and used this configuration:
```yaml
rule:
  global:
    indentSize : 4
    blank_line_ends_group : False
    comment_line_ends_group : False
    compact_alignment : True
    case_control_statements_end_group : False
    style : 'no_blank_line'
    if_control_statements_ends_group : False
  architecture_026:
    disable : True
  signal_007:
    disable : True
  architecture_029 :
    disable : True
  constant_101:
    disable : False
  variable_100:
    disable : False
  signal_100:
    disable : False
  alias_declaration_103:
    disable : False
  file_100:
    disable : False
  subtype_100:
    disable : False
  type_100:
    disable : False
```
I had to disable `signal_007` as it was complaining about the assignments on the signal declarations.

The rules `constant_101` through `type_100` are new and disabled by default.  They are configured to enforce a single space before the identifier.  They are disabled because they will conflict with `architecture_029`.

Applying the above configuration to your file resulted in the following output:
```vhdl
architecture rtl of sig_type_const is
    constant this_is_a_constant_0 : natural := 599; -- comment0
    constant this_is_a_constant_1 : natural := 599; -- comment1
    constant this_is_a_constant_2 : natural := 599; -- comment2
    type fsm_type_0 is (idle, do_something, done);  -- comment
    type fsm_type_0 is (idle, do_something, done);  -- comment
    type fsm_type_0 is (idle, do_something, done);  -- comment
    signal state_0 : fsm_type := idle;              -- comment
    signal state_1 : fsm_type := idle;              -- comment
    signal state_2 : fsm_type := idle;              -- comment
begin
end architecture rtl;
```

I pushed an update to the `issue-815` branch.  When you get a chance could you check it out and let me know if it address the issue.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-10-12:
Morning @njm295 ,

There are rules which enforce whitespace between these tokens and the number of spaces can be configured.  In this case the rules `constant_006`, `constant_010` and `signal_006` are applicable.  I currently to not have a rule for the default assignment character for signal declarations, but I can add one.

Try this updated configuration:
```yaml
rule:
  global:
    indentSize : 4
    blank_line_ends_group : False
    comment_line_ends_group : False
    compact_alignment : True
    case_control_statements_end_group : False
    style : 'no_blank_line'
    if_control_statements_ends_group : False
  architecture_026:
    disable : True
  constant_006:
    number_of_spaces: 1
  constant_010:
    number_of_spaces: 1
  signal_006:
    number_of_spaces: 1
  signal_007:
    disable : True
  architecture_029 :
    disable : True
  constant_101:
    disable : False
  variable_100:
    disable : False
  signal_100:
    disable : False
  alias_declaration_103:
    disable : False
  file_100:
    disable : False
  subtype_100:
    disable : False
  type_100:
    disable : False
```
It should get you 99% there.

I will add the new rule tonight.

Regards,

--Jeremy

njm295, 2022-10-12:
Hi Jeremy,
Tried the constant_(006,010) and singal_006.  They did as advertised.  Looking forward to the default_signal_assignment rule!

Thanks for the support on this.

nick

jeremiah-c-leary, 2022-10-12:
Hey @njm295 ,

I pushed an update where I added rules `signal_100` and `signal_101`.  They deal  with whitespace before and after the assignment operator `:=`.

If you use the following configuration, you should get the formatting you want:
```yaml
rule:
  global:
    indentSize : 4
    blank_line_ends_group : False
    comment_line_ends_group : False
    compact_alignment : True
    case_control_statements_end_group : False
    style : 'no_blank_line'
    if_control_statements_ends_group : False
  architecture_026:
    disable : True
  constant_006:
    number_of_spaces: 1
  constant_010:
    number_of_spaces: 1
  signal_006:
    number_of_spaces: 1
  signal_101:
    number_of_spaces: 1
  signal_007:
    disable : True
  architecture_029 :
    disable : True
  constant_101:
    disable : False
  variable_100:
    disable : False
  signal_100:
    disable : False
  alias_declaration_103:
    disable : False
  file_100:
    disable : False
  subtype_100:
    disable : False
  type_100:
    disable : False
```

--Jeremy

jeremiah-c-leary, 2022-10-18:
Hey @njm295 ,

Just wanted to ping you to see if you had a chance to check out the latest update.

Thanks,

--Jeremy

njm295, 2022-10-18:
Hi Jeremy,
I did try the new rules and suggested configuration.  I don't think the "number of spaces solution" will work.   I added a few lines to stress the formatting.  The following screenshot shows the result.  I think constants, signals, type_defs, aliases, ... should be treated as a blocks with formatting/alignment/blank_lines(above,within,below)  configured uniquely for each block.
[image]

jeremiah-c-leary, 2022-10-18:
Hey @njm295 ,

Just to be clear, is this the way you would like the code formatted:

```vhdl
    constant this_is_a_constant_0        : natural                      := 599;  -- comment0
    constant this_is_a_constant_1        : natural                      := 599;  -- comment1
    constant this_is_a_constant_2        : natural                      := 599;  -- comment2
    constant this_is_a_bigger_constant_2 : std_logic_vector(7 downto 0) := x"0"; -- comment
    type fsm_type_0        is (idle, do_something, done);                        -- comment
    type fsm_type_0        is (idle, do_something, done);                        -- comment
    type fsm_type_0        is (idle, do_something, done);                        -- comment
    type bigger_fsm_type_2 is (idle, do_something, done);                        -- comment
    signal state_0                         : fsm_type             := idle;       -- comment
    signal state_1                         : fsm_type             := idle;       -- comment
    signal state_2                         : fsm_type             := idle;       -- comment
    signal very_large_signal_name_to_check : some_long_named_type := idle;       -- comment
```

--Jeremy

njm295, 2022-10-18:
Yes, and also to be able to have blank lines separating each group.

njm295, 2022-10-18:
> Is this the final formatting:

Yes!

jeremiah-c-leary, 2022-10-19:
Hey @njm295 ,

I pushed another update to the `issue-815` branch.

Try this configuration:
```yaml
rule:
  global:
    indentSize : 4
    blank_line_ends_group : True
    comment_line_ends_group : False
    case_control_statements_end_group : False
    if_control_statements_ends_group : False
  architecture_027:
    blank_line_ends_group : False
  constant_010:
    number_of_spaces: '>=1'
  signal_101:
    number_of_spaces: '>=1'
  signal_007:
    disable : True
  architecture_029 :
    disable : True
  constant_101:
    disable : False
  variable_100:
    disable : False
  signal_100:
    disable : False
  alias_declaration_103:
    disable : False
  file_100:
    disable : False
  subtype_100:
    disable : False
  type_100:
    disable : False
  constant_200:
    disable : False
  type_006:
    number_of_spaces: '>=1'
  type_010:
    disable : True
  type_011:
    disable : True
  type_200:
    disable : False
  signal_200:
    disable : False
```

I am not 100% about some of changes I made, but let me know if there are any other issues with this.  There will need to be some changes to this before I would be willing to merge it to master.

Regards,

--Jeremy

njm295, 2022-10-21:
Hi Jeremy
I tried  the testcase with your configuration as a yaml file and the 
output looked good.
When I try the configuration using a json configuration file  I get rule 
conflicts!

I generated the oc file for each run above and they were indeed different!

Thanks,
nick

jeremiah-c-leary, 2022-10-21:
Hey @njm295 ,

> I generated the oc file for each run above and they were indeed different!

That is unexpected.  I will try it on my end.

--Jeremy

jeremiah-c-leary, 2022-10-21:
Hey @njm295 ,

I tried this configuration:
```yaml
rule:
  global:
    indentSize : 4
    blank_line_ends_group : True
    comment_line_ends_group : False
    case_control_statements_end_group : False
    if_control_statements_ends_group : False
  architecture_027:
    blank_line_ends_group : False
  constant_010:
    number_of_spaces: '>=1'
  signal_101:
    number_of_spaces: '>=1'
  signal_007:
    disable : True
  architecture_029 :
    disable : True
  constant_101:
    disable : False
  variable_100:
    disable : False
  signal_100:
    disable : False
  alias_declaration_103:
    disable : False
  file_100:
    disable : False
  subtype_100:
    disable : False
  type_100:
    disable : False
  constant_200:
    disable : False
  type_006:
    number_of_spaces: '>=1'
  type_010:
    disable : True
  type_011:
    disable : True
  type_200:
    disable : False
  signal_200:
    disable : False
```
and converted it to json:
``` json
{
  "rule": {
    "global": {
      "indentSize": 4,
      "blank_line_ends_group": true,
      "comment_line_ends_group": false,
      "case_control_statements_end_group": false,
      "if_control_statements_ends_group": false
    },
    "architecture_027": {
      "blank_line_ends_group": false
    },
    "constant_010": {
      "number_of_spaces": ">=1"
    },
    "signal_101": {
      "number_of_spaces": ">=1"
    },
    "signal_007": {
      "disable": true
    },
    "architecture_029": {
      "disable": true
    },
    "constant_101": {
      "disable": false
    },
    "variable_100": {
      "disable": false
    },
    "signal_100": {
      "disable": false
    },
    "alias_declaration_103": {
      "disable": false
    },
    "file_100": {
      "disable": false
    },
    "subtype_100": {
      "disable": false
    },
    "type_100": {
      "disable": false
    },
    "constant_200": {
      "disable": false
    },
    "type_006": {
      "number_of_spaces": ">=1"
    },
    "type_010": {
      "disable": true
    },
    "type_011": {
      "disable": true
    },
    "type_200": {
      "disable": false
    },
    "signal_200": {
      "disable": false
    }
  }
}
```
I used the `-oc` option and created two json files:
```mono
$ vsg -c config.yaml -oc yaml.json
$ vsg -c config.json -oc json.json
```
Then used md5sum to check them:
```mono
$ md5sum yaml.json json.json
121f001aa7c1c9a838b9fd1218a918b6  yaml.json
121f001aa7c1c9a838b9fd1218a918b6  json.json
```
which shows that mine matched.

Could you double check your json file against the yaml file?

Thanks,

--Jeremy

njm295, 2022-10-21:
You are correct,   Pilot error on my part.
these configurations give the formatting that I request.

I noticed that  when I invoke vsg as follows:
vsg -f sig_type_const.vhd -b --fix --configuration ~/vsgTest/myRules.yaml -oc rules.yaml

It only ouputs the rules.yaml file, not the fixed nor the .bak file... is this expected?

jeremiah-c-leary, 2022-10-23:
Hey @njm295 ,

That was the intended behavior.  The same behavior applies to the `-rc` option also.  I had not learned about subcommands in the argument parser when I started.

--Jeremy

jeremiah-c-leary, 2022-11-07:
Morning @njm295,

Could you let me know where we stand on this issue?

Thanks,

--Jeremy

njm295, 2022-11-07:
Good morning Jeremy,
The updates   give me the formatting I desire.  
Thanks you for the support!
-- nick

jeremiah-c-leary, 2022-12-01:
Hey @njm295 ,

I need to re-open this issue due a structural change in VSG.  After the change has been made I will re-visit this so I can get the changes merged to master.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-11-29:
Good Afternoon @njm295 ,

I merged master into this branch and made a couple of updates:

  1)  Removed architecture_401 (I duplicated it on a previous PR)
  2)  Added a configuration option to `architecture_026` to allow for alignment of the type is keyword.

When you get a chance could you check the branch out again and let me know if it still works for you?

Thanks,

--Jeremy

njm295, 2023-12-02:
HI Jeremiah
It's been a while...   I checked out  branch issue-815 and tested.  I think we are close.  I attach the orig vhdl file, the fixed file and the rules file.  
Constant names needs alignment fix
type needs alignment fix to the is keyword

There may me some rules to fix this; but I do not know how to read the documentation for  branch?

[myRules.yaml.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/13536653/myRules.yaml.txt)
[sig_type_const.vhd.orig.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/13536654/sig_type_const.vhd.orig.txt)
[sig_type_const.vhd.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/13536655/sig_type_const.vhd.txt)

jeremiah-c-leary, 2023-12-02:
Good Morning @njm295 ,

Yeah, it has been awhile.

> There may me some rules to fix this; but I do not know how to read the documentation updates for checked out branch?

If you navigate to the `docs` directory you can issue the following command `make html`.  This will generate all the documentation in the `docs/_build` directory.  Load the `index.html` into your browser and you should be good to go.

> Constant names needs alignment fix
type needs alignment fix to the is keyword

I updated your config file:
```yaml
  1 rule:
  2   global:
  3     indentSize : 4
  4     blank_line_ends_group : False
  5     comment_line_ends_group : False
  6     compact_alignment : True
  7     case_control_statements_end_group : False
  8     style : 'no_blank_line'
  9     if_control_statements_ends_group : False
 10   signal_007:
 11     disable : True
 12   architecture_026 :
 13     include_type_is_keyword : 'yes'
 14   architecture_029 :
 15     disable : True
 16   variable_100:
 17     disable : False
 18   signal_100:
 19     disable : False
 20   alias_declaration_103:
 21     disable : False
 22   file_100:
 23     disable : False
 24   subtype_100:
 25     disable : False
 26   type_100:
 27     disable : False
 28   constant_101:
 29     disable : False
 30   type_006:
 31     number_of_spaces : '>=1'
```
I changed line 13 from `no` to `'yes'` and added lines 28 through 31.

> Also, would like to remove extra leading spaces in the comment statement.

There is already rule `comment_100` which will remove spaces after the `--` in comments.  I am not sure why that is not working.   I will investigate.

Could try the changes to the config file above while I debug why `comment_100` is not working?

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-02:
So the rule `comment_100` just ensures there is a space after the comment `--`.  I would need to create a new rule to ensure there is no more than one space after the `--`.  

However, a rule to remove the spaces would prevent any kind of indenting in comments.

Let me know if you would like me to implement this rule.

--Jeremy

njm295, 2023-12-02:
> Good Morning @njm295 ,
>
> Yeah, it has been awhile.
>
>
> If you navigate to the `docs` directory you can issue the following command `make html`. This will generate all the documentation in the [...]

I tried the proposed changes to the config file:  Here's screenshot of the results:
[image]

Why is the 'is' keyword in the type line indented to the same level as the : in previous lines?

jeremiah-c-leary, 2023-12-02:
Afternoon @njm295 ,

Hmm....that was line 13 being set to yes.  Looking back at your original request I can see that you wanted the `is` keyword aligned for types but not aligned with the colons for constants etc...

Can you try the following configuration:
```yaml
  1 rule:
  2   global:
  3     indentSize : 4
  4     blank_line_ends_group : True
  5     comment_line_ends_group : False
  6     case_control_statements_end_group : False
  7     if_control_statements_ends_group : False
  8   architecture_026 :
  9     include_type_is_keyword : 'yes'
 10   architecture_027:
 11     blank_line_ends_group : False
 12   constant_010:
 13     number_of_spaces: '>=1'
 14   signal_101:
 15     number_of_spaces: '>=1'
 16   signal_007:
 17     disable : True
 18   architecture_029 :
 19     disable : True
 20   constant_101:
 21     disable : False
 22   variable_100:
 23     disable : False
 24   signal_100:
 25     disable : False
 26   alias_declaration_103:
 27     disable : False
 28   file_100:
 29     disable : False
 30   subtype_100:
 31     disable : False
 32   type_100:
 33     disable : False
 34   constant_200:
 35     disable : False
 36   type_006:
 37     number_of_spaces: '>=1'
 38   type_010:
 39     disable : True
 40   type_011:
 41     disable : True
 42   type_200:
 43     disable : False
 44   signal_200:
 45     disable : False
```
When I use this against your code snippet, it is formatted like this:
```vhdl
  1 library ieee;
  2     use ieee.std_logic_1164.all;
  3
  4 entity sig_type_const is
  5     port (
  6         clk   : in    std_logic;
  7         reset : in    std_logic
  8     );
  9 end entity sig_type_const;
 10
 11 architecture rtl of sig_type_const is
 12
 13     constant this_is_a_constant_0 : natural := 599;   -- comment0
 14     constant this_is_a_constant_1 : natural := 599;   -- comment1
 15     constant this_is_a_constant_2 : natural := 599;   -- comment2
 16
 17     type fsm_type_0    is (idle, do_something, done); -- comment
 18     type fsm_type_01   is (idle, do_something, done); -- comment
 19     type fsm_type_0123 is (idle, do_something, done); -- comment
 20
 21     signal state_0   : fsm_type := idle;              -- comment
 22     signal state_12  : fsm_type := idle;              --  comment
 23     signal state_123 : fsm_type := idle;              --   comment
 24
 25 begin
 26
 27 end architecture rtl;
 28
```
which I believe is your desired format.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-09:
Morning @njm295 ,

I will get this merged to master.

--Jeremy

PR #840 (2023-12-09): Issue 815
Resolves #815
closed 2023-12-09 by jeremiah-c-leary