jeremiah-c-leary/vhdl-style-guide #816: Modification for signal_008 to allow pipeline numbers

SethGower, 2022-09-07:
**Is your feature request related to a problem? Please describe.**
No, it is not a problem, just a possible enhancement. I would like to be able to have signal names that have a prefix, using [`signal_008`](https://vhdl-style-guide.readthedocs.io/en/latest/signal_rules.html#signal-008), but allow for pipeline stage signal prefixes as well. 

**Describe the solution you'd like**
If a signal is the output of the first register of a pipeline, then allow the signal name to be `s1_signal_name`. Currently since the prefix doesn't match exactly what is set (for example `s_`) then this is counted as an error. 

**Describe alternatives you've considered**
Obviously one could do `s_signal_1` instead, but I find it is easier to understand faster if it can be `s1_signal_name`.

jeremiah-c-leary, 2022-09-22:
Hey @SethGower,

One possible solution would to be add the delay prefixes into a configuration:
```json
{
  "rule": {
    "signal_008": {
      "prefixes": [
        "s_", "s1_", "s2_", "s3_", "s4_"
      ]
    }
  }
}
```
You would have to ensure the number is high enough for your delays.

Would that work for you?

Regards,

--Jeremy

SethGower, 2022-09-26:
That's a good point, and a good solution. I was thinking of a more general solution, but realistically if you need a ton of stages, it's best to use a vector (in my opinion). Thanks!

PS: This is a pretty awesome project, thanks for the work you've done!

jeremiah-c-leary, 2022-09-26:
Hey @SethGower,

> That's a good point, and a good solution. I was thinking of a more general solution, but realistically if you need a ton of stages, it's best to use a vector (in my opinion). Thanks!

Are you okay with closing this issue?

> PS: This is a pretty awesome project, thanks for the work you've done!

Thanks for the kind words.  It does a mean a lot.

Regards,

--Jeremy