jeremiah-c-leary/vhdl-style-guide #817: Problem parsing integer range declaration in generic

elpollodiablo84, 2022-09-12:
There's some strange behavior with generics and range declaration in a component (I'm on `vsg` version 3.12.0).
This is parsed with 0 errors:
```
entity test is
end entity test;

architecture rtl of test is

  component test3 is
    generic (
      g_test2 : integer
    );
    port (
      clk_i     : in    std_logic;
      reset_n_i : in    std_logic;

      test_i           : in    std_logic;
      test_test_test_o : out   std_logic
    );
  end component test3;

begin

end architecture rtl;
```
Specifying the integer range generates some problems:
```
entity test is
end entity test;

architecture rtl of test is

  component test3 is
    generic (
      g_test2 : integer range 0 to 3
    );
    port (
      clk_i     : in    std_logic;
      reset_n_i : in    std_logic;

      test_i           : in    std_logic;
      test_test_test_o : out   std_logic
    );
  end component test3;

begin

end architecture rtl;
```
Output of `vsg -f test.vhd -ap`:
```
Total Rules Checked: 663
Total Violations:    3
  Error   :     3
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  component_017             | Error      |          8 | Move : 2 columns
  generic_004               | Error      |         10 | Indent level 3
  generic_007               | Error      |         10 | Change "(" to "("
----------------------------+------------+------------+--------------------------------------
```

jeremiah-c-leary, 2022-09-13:
Hey @elpollodiablo84,

I pushed a fix for this to the `issue-817` branch.  When you get a chance could you check it out and let me know if it fixes the issue on your side.

Thanks,

--Jeremy

elpollodiablo84, 2022-09-14:
I can confirm that your fix works as intended (on my code also). Thank you!

jeremiah-c-leary, 2022-09-14:
Hey @elpollodiablo84,

Awesome.  I will merge this to master and release 3.12.1.

--Jeremy

PR #818 (2022-09-14): Updates for issue #817:
Fixed issue with range_constraint classification not stopping when encountering an unmatched closing parenthesis.

  1)  Updated test
  2)  Updated range_constraint classification

Resolves #817
closed 2022-09-14 by jeremiah-c-leary