jeremiah-c-leary/vhdl-style-guide #819: Parenthesis in range definition results in error

Scafir, 2022-09-19:
**Environment**
CentOS Linux 7 (Core)
VHDL Style Guide (VSG) version: 3.12.1+zip.file

**Describe the bug**
If any of the values defining the range's bounds is between parentheses, the vsg fails with error `Error: Unexpected token detected while parsing XXX @ Line X, Column XX in file None`. This can be a bit nasty to discover because when linting multiple files, the error is lost in the logs, but may cause failure elsewhere (e.g: junit file generation).

**To Reproduce**
Steps to reproduce the behavior:
1. Create a sample file, e.g:
```
entity sample is
  port (
  );
end entity sample;

architecture behavioral of sample is

  signal a : integer range (10) downto 4;

begin

end architecture behavioural;
```
2. run `vsg -f sample.vhd`
3. See error

**Expected behavior**
The parentheses should be ignored

**End**
Thank you very much for this tool!

Scafir, 2022-09-20:
Note that the parentheses do not need to fully enclose the bound. E.g: 
```
generic(a  : natural range 1 to (myConstant * 1_000) / 8 := 1);
```
will also result in errors.

jeremiah-c-leary, 2022-09-22:
Hey Scafir,

I pushed an update for this issue to the `issue-819` branch.  When you get a chance could you validate it works on your end.

> Thank you very much for this tool!

Thanks for the kind words.

Regards,

--Jeremy

Scafir, 2022-09-22:
Hi Jeremiah,
I can't believe that you already fixed my issues!
I confirm that that `issue-819` indeed fix the issue without impacting the rest of the operations on my codebase :)

PR #822 (2022-09-22): Updates for issue #819
Constraints with parenthesis would not properly classify.

  1)  Updated test to expose issue
  2)  Updated constraint classifer to properly handle parenthesis

Resolves #819
closed 2022-09-22 by jeremiah-c-leary