jeremiah-c-leary/vhdl-style-guide #820: generate_002 does not like 0 whitespaces

Scafir, 2022-09-20:
**Environment**
CentOS Linux 7 (Core)
VHDL Style Guide (VSG) version: 3.12.1+zip.file

**Describe the bug**
When using the rule
```
rule:
  generate_002:
    number_of_spaces: 0
```
`vsg` will complain that
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  generate_002              | Error      |        XXX | Add 0 spaces between LABEL and :
```
**To Reproduce**
Steps to reproduce the behavior:
1. Use 0 spaces between colon and label in a generate statement, e.g:
```
architecture behavioral of sample is
    signal a : std_logic;
begin
      MY_LABEL: if a = '0' generate
        a <= '1';
    end generate MY_LABEL;
end architecture behavioral;
```
2. Run `vsg -c generate002_0space.yaml --fix -f sample.vhd`
3. See error

**Expected behavior**
0 spaces between colon and label should be considered correct if number_of_spaces is set to 0.

jeremiah-c-leary, 2022-09-22:
Hey @Scafir,

I pushed an update to the `issue-820` branch.  When you get a chance could you check it out on your end.

Thanks,

--Jeremy

Scafir, 2022-09-22:
Hi @jeremiah-c-leary,
The `issue-820` branch seem to do the trick  without any side effects.

jeremiah-c-leary, 2022-09-22:
Awesome, I will merge this to master also.

--Jeremy

PR #823 (2022-09-22): Updates for issue #820:
Configuring whitespace to 0 would not enforce 0 whitespace.

  1)  Updated tests
  2)  Updated whitespace_between_tokens base rule

Resolves #820
closed 2022-09-22 by jeremiah-c-leary