jeremiah-c-leary/vhdl-style-guide #821: when_001 fails to fix if there is no space between boolean and else clause

Scafir, 2022-09-20:
**Environment**
CentOS Linux 7 (Core)
VHDL Style Guide (VSG) version: 3.12.1+zip.file

**Describe the bug**
If there is no whitespace between the boolean test and `else` keyword in a `when` statement, an incorrect solution is suggested when running `vsg` and a crash occurs when running `vsg --fix`.
The lack of whitespace can occur in two instances (that I could come up with):
```
a <= b when (c = '1')else d;
```
```
a <= b when c = '1'else d;
```
I suppose internally, it has either something to do with delimiters, or bad indexing.

**Error message**
The resulting error message (when running with `--fix`):
```
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/REDACTED/.local/lib/python3.6/site-packages/vsg/rules/utils.py", line 61, in insert_token
    oToken = update_code_tags(lTokens[index], oToken)
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib64/python3.6/multiprocessing/pool.py", line 119, in worker
    result = (True, func(*args, **kwds))
  File "/REDACTED/python3.6/site-packages/vsg/apply_rules.py", line 103, in apply_rules
    commandLineArguments.fix_phase, commandLineArguments.skip_phase, fix_only
  File "/REDACTED/.local/lib/python3.6/site-packages/vsg/rule_list.py", line 158, in fix
    oRule.fix(self.oVhdlFile, dFixOnly)
  File "/REDACTED/.local/lib/python3.6/site-packages/vsg/rule.py", line 106, in fix
    self._fix_violation(oViolation)
  File "/REDACTED/.local/lib/python3.6/site-packages/vsg/rules/when/rule_001.py", line 89, in _fix_violation
    rules_utils.insert_token(lTokens, 0, oToken)
  File "/REDACTED/.local/lib/python3.6/site-packages/vsg/rules/utils.py", line 63, in insert_token
    oToken = update_code_tags(lTokens[0], oToken)
IndexError: list index out of range
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/REDACTED/.local/bin/vsg", line 11, in <module>
    load_entry_point('vsg==3.12.1', 'console_scripts', 'vsg')()
  File "/REDACTED/.local/lib/python3.6/site-packages/vsg/__main__.py", line 140, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/usr/lib64/python3.6/multiprocessing/pool.py", line 735, in next
    raise value
IndexError: list index out of range
```
**To Reproduce**
1. Use 0 spaces in a file. E.g:
```
architecture behavioral of sample is
begin
    a <= b when (c = '1')else d;
end architecture behavioral
```
2. Run `vsg --fix -f sample.vhd`
3. See error

**Expected behavior**
The lack of whitespace should be detected as such and fixed by adding one.

jeremiah-c-leary, 2022-09-22:
Hey @Scafir,

I pushed an update for this issue to the `issue-821` branch.  When you get a chance could you check it out on your end.

Thanks,

--Jeremy

Scafir, 2022-09-22:
Hi @jeremiah-c-leary ,
The fix does not seem to work. On one of my production files, the issue is ignored (`vsg` does not recognize lack of whitespace, but does not crash), but more importantly, the sample file suggested in the initial comment crashes with:
```
Error: Unexpected token detected while parsing architecture_body @ Line 4, Column 1 in file None
       Expecting : ;
       Found     : 
```
Looking back on my initial bug report, I wonder whether a extra `when` rule should be added, instead of  `when_001` managing both `else` line break and spaces..

jeremiah-c-leary, 2022-09-22:
Hey @Scafir,

>  ...the sample file suggested in the initial comment crashes with:

When I run this example:
```vhdl
architecture rtl of fifo is
begin

  a <= b when (c = '1')else d;
  a <= b when c = '1'else d;

end architecture rtl;
```
I do not get a parsing error.

Would it be possible to provide the file itself, or a pared down version of the file, that shows the issue.

> Looking back on my initial bug report, I wonder whether a extra when rule should be added, instead of when_001 managing both else line break and spaces..

Hmm...there is no rule that enforces spaces before the `else` keyword.  If you would like one let me know.

Regards,

--Jeremy

Scafir, 2022-09-22:
Please excuse me! I forgot to add  the final `;`  after `end architecture behavioral` in the initial comment. This was causing the issue and not the modifications you made. (though the fact that the file name reported `None` confused me a bit).
 
Concerning the rules, I was mainly referring to the "Expected Behavior" of my intial comment, which does not really make sense (the true issue that I had was the crash, not the lack of space).
I believe that it would make sense to add a rule checking the presence of this space, as I find its absence a bit ugly as well as giving unnecessary wiggle room. (though I don't want to make this sound like a request, I am already over the moon with this software!)

With this last fix merged, `vsg` will perform ideally on my codebase of 15'000 odd lines, I really want to thank you for all the work you put in for this tool!

PR #824 (2022-09-23): Updates for issue #821
The else keyword in the when statement without a space before it would fail when_001.

  1)  Added tests to expose issue
  2)  Updated rule when_001 to handle no spaces before else keyword

Resolves #821
closed 2022-09-23 by jeremiah-c-leary