jeremiah-c-leary/vhdl-style-guide #827: comment_100: Non alpha numeric characters starting comment

Scafir, 2022-09-30:
I would not know if this is a bug report or a feature request  (in any case, I am slowly getting closer to have my codebase tamed :) 

### Environment
CentOS Linux 7 (Core)
VHDL Style Guide (VSG) version: 3.12.2+zip.file

### Describe the bug
If a comment starts with a non-alphanumeric character and the first space is missing, the suggested fix inserts a whitespace _after_ the non-alpha numeric character.
E.g: `--(2+1)*4 is the solution` becomes:  `--( 2+1)*4 is the solution` instead of  `-- (2+1)*4 is the solution`. 
I guess that this behavior can be explained due to documentation frameworks requiring the very first character of a comment  to be something specific (typically `--!` for doxygen). 

### To Reproduce
1. Use any non-alphanumeric character as following a comment (without whitespace)
2. Run vsg with comment_100 enabled on the file

### Expected behavior
It could be good to either:
1) Mention this in the documentation as a "deal with it".
2) Insert a space right after `--` in all cases except when the first character is `!`.
3) Enable the user to chose which characters should be ignored.

I would personally go for number 2.

jeremiah-c-leary, 2022-10-02:
Hey @Scafir,

Rule `comment_100` partially exists to support block comments.

> It could be good to either:
>
> 1. Mention this in the documentation as a "deal with it".
> 2. Insert a space right after -- in all cases except when the first character is !.
> 3. Enable the user to chose which characters should be ignored.

I would lean towards option 3 and add `!` as one of the defaults.  That would give the user the most flexibility.

Sound reasonable?

Regards,

--Jeremy

Scafir, 2022-10-03:
Hi @jeremiah-c-leary ,
If option 3 is not too cumbersome to implement, it is by far the most "perfect" solution. It sounds reasonable to me!
Best Regards,

jeremiah-c-leary, 2022-10-08:
Hey @Scafir ,

I pushed an update to this to the `issue-827` branch.

I added the ability to define your own exceptions using the **exception** option.

When you get a chance could you give it a try.

Thanks,

--Jeremy

Scafir, 2022-10-10:
Hi @jeremiah-c-leary ,

Hope you had a nice week end!

I am not quite sure how to use the exception, it it via a list with this formatting: `exception: [ "(" , "!" , ...]`? If yes, it does not seem to be working (or I am doing something wrong).
Also it seems that by default, it is ignoring "!", which is nice. Would it be possible to remove the exception on "!" by specifying an empty list? I don't really see the case where this would be useful, but it is still interesting to see that documented.

It seems like hitting an exception stops checking: `--!test` becomes `--!test` instead of `--! test`. I would prefer that the `!` to be considered as part of the comment block and thus get `--! test`.

jeremiah-c-leary, 2022-10-10:
Hey @Scafir ,

> Hope you had a nice week end!

It was pretty good.  I am driving to finish off the last remaining issues before my next release.

> I am not quite sure how to use the exception, it it via a list with this formatting:

You can always check the options for a rule using the `-rc` command line option.  For this rule it looks like this:
```mono
$ vsg -rc comment_100
{
  "rule": {
    "comment_100": {
      "indentSize": 2,
      "phase": 2,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "exceptions": [
        "--!",
        "--=",
        "--+",
        "--|",
        "---"
      ]
    }
  }
}
```

I had wondered about using `--!` versus `!`.  However, it may make more sense to use just `!`.  What are your thoughts?

> Would it be possible to remove the exception on "!" by specifying an empty list?

I can confirm using an empty list for `exceptions` will catch `--!`
```mono
================================================================================
File:  original.vhd
================================================================================
Phase 2 of 7... Reporting
Total Rules Checked: 250
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  comment_100               | Error      |          2 | Change "--!" to "-- !"
----------------------------+------------+------------+--------------------------------------
```
> It seems like hitting an exception stops checking: --!test becomes --!test instead of --! test. I would prefer that the ! to be considered as part of the comment block and thus get --! test.

I could add another rule to enforce whitespace after a comment pattern.  Then the two rules work together to enforce whitespacing.

Regards,

--Jeremy

jeremiah-c-leary, 2022-10-10:
Hey @Scafir ,

I pushed an update to this issue where I added the rule `comment_101`.  This rule will add a whitespace after a user defined pattern.
```json
{
  "rule": {
    "comment_101": {
      "indentSize": 2,
      "phase": 2,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "patterns": [
        "--!",
        "--|"
      ]
    }
  }
}
```
Give it a try and let me know what you think.

Regards,

--Jeremy

Scafir, 2022-10-10:
Hi @jeremiah-c-leary ,
Looking forward to the new update :)

> You can always check the options for a rule using the -rc command line option. For this rule it looks like this:

I must have skimmed over this option too fast when reading the help output! Thanks for the trick.

> I had wondered about using --! versus !. However, it may make more sense to use just !. What are your thoughts?

Initially, I would have been in favor of using `!` versus `--!`. However, I now believe that your approach makes the most sense. Even if programmatically  speaking we want to "avoid" a particular character, the true goal is really to make `vsg` understand that `--!` is a special type of comment and should be handled as such. I wonder if it would be possible to abuse this system in some way by using something which does not start by `--` (I have not found a way).

I am happy with how the changes behave. However, I am not quite sure I understand why there is now two rules. This means that the user has to tell `vsg` twice about its intent to have a special type of comment. Would it be possible to merge them?

Best Regards

Scafir, 2022-10-10:
Hi @jeremiah-c-leary,

I feel your struggle, #837 is making my head hurt as soon as I start thinking about it again. Merging the two of them seems like the best way to me on this one though!
Thanks again for the care you take on this project :)

jeremiah-c-leary, 2022-10-10:
Evening @Scafir ,

I merged the `comment_101` into `comment_100`.   Let me know what you think.

> Thanks again for the care you take on this project :)

Thanks for the kind words.

--Jeremy

Scafir, 2022-10-11:
Hi @jeremiah-c-leary ,

When using the defaults, everything seems to work OK, however It seems there are some issues if I use the following rules:
```
rule:
  comment_100:
    exceptions: ["--%"]
```
Using the following sample file:
```
--SimpleComment
--!Doxy
--%Yes
```
`vsg` fails to fix the file in one go, and will wants to add a space both before and after `!`. It also does not add a space after `%`.

What makes situation more complicated is that in some instances some people do not want a space after the comment block (e.g when delimiting sections like: `------------------- Registers -------------`). Maybe having specifying parameters as dual (`exceptions: [ ("--!", 1), ("--", 0) , ...`) can satisfy every usecase?

Regards

jeremiah-c-leary, 2022-10-18:
Evening @Scafir ,

I was wondering if you had a chance to try the configuration above?  I am hoping to do a release this weekend if a couple of other issues get closed and would like to include this.

--Jeremy

jeremiah-c-leary, 2022-11-07:
Morning @Scafir,

Just wanted to ping you to see if you had a change to validate the update.

Thanks,

--Jeremy

Scafir, 2022-11-08:
Hi @jeremiah-c-leary ,

Sorry for the delay, holidays came and I got distracted.

It seems that `patterns` are not listed when running `vsg -rc comment_100`, which makes testing a bit harder. Otherwise, it seems to perform as intended.

Best Regards,
Clyde

jeremiah-c-leary, 2022-12-02:
Hey @Scafir ,

I managed to lose track of this issue.

I added the `patterns` to the configuration list so now it will show up as an option:

```text
$ vsg -rc comment_100
{
  "rule": {
    "comment_100": {
      "indentSize": 2,
      "phase": 2,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "exceptions": [
        "--!",
        "--=",
        "--+",
        "--|",
        "---"
      ],
      "patterns": [
        "--!",
        "--|"
      ]
    }
  }
}
```

Let me know how it works out.

--Jeremy

Scafir, 2022-12-09:
Hey @jeremiah-c-leary ,

Third time the charm! I'm happy with how it is working.
I suggest a small change to [documentation](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-827/docs/configuring_whitespace_after_comment_rules.rst) to:

exceptions: A list of patterns to not insert whitespace after "`--`".    
patterns:  A list of patterns to insert whitespace after the specified pattern.

I would also mention that patterns and exceptions not starting with "--" will get silently ignored.

Regards,
Clyde

jeremiah-c-leary, 2022-12-09:
Morning @Scafir ,

I updated the documentation to match my current style.

Let me know what you think.

Regards,

--Jeremy

Scafir, 2022-12-12:
Hi @jeremiah-c-leary ,

I hope you had a good weekend.

I believe that there are still possible improvements to the documentation:

In the table, line patterns, it is not clear that the whitespace will be inserted after the whole pattern and not `--`. Now that "exceptions" is fixed, I think that the older formulation (A list of patterns to insert whitespace after.) is a bit clearer. 

The "Note" styling is a bit strange.. I would put it higher in the hierarchy.

The first and second line under "Note" are contradicting each other:
> Each exception and pattern must [...] not more than three characters long.
> Each pattern [...] there are no restrictions on length.

I would rephrase it like that:
Each `exception` must start with `--` and be no more than three characters long.
Each `pattern` must start with `--`.  There are no restrictions on length.

Regards,
Clyde

jeremiah-c-leary, 2022-12-12:
Morning @Scafir ,

I tried moving the information in the Note into the description for each option.  Let me know what you think.

Thanks,

--Jeremy

Scafir, 2022-12-13:
Hi @jeremiah-c-leary ,

There should be an upper-case "E" in` [...] >e<ach pattern must start with`. Apart from this, I am happy with the changes! The examples makes the rules crystal clear (I apparently misunderstood the `patterns` in my last comment, sorry for the confusion).
Feel free to close this issue once it is merged :)

Regards,
Clyde

jeremiah-c-leary, 2022-12-14:
Evening @Scafir ,

I updated the E and will merge this to master.

--Jeremy

PR #838 (2022-12-14): Updates for issue #827
Changing behavior of comment_100:

  1)  block comments are ignored
  2)  Exceptions added to ignore rule

Resolves #827
closed 2022-12-14 by jeremiah-c-leary