jeremiah-c-leary/vhdl-style-guide #829: If two port_026 trigger at the same time with --fix, entity_017 fails

Scafir, 2022-10-03:
**Environment**
CentOS Linux 7 (Core)
VHDL Style Guide (VSG) version: 3.12.2+zip.file

**Describe the bug**
If two consecutive port lines must be broken up because of multiple signals declaration (`port_026`, phase 1) with names of different lengths, the suggested fix is not aligned. This in itself is not a big issue, but the fixer then fails at solving `entity_017`, ensuring alignment. I believe that I see three (four?) intertwined bugs here:
1. Bad alignment in solving `port_026`.
2. `entity_017` reports negatives offsets (wants to erase part of the longest variable name), instead of moving the shorter declarations to match the longest one.
3. For some reason, running `vsg --fix ...` a second time fixes the files successfully. Which may indicate that the run of `port_026` (phase 1) and `entity_017` (phase 5) are not independent.
4. For some reason, the first run of `vsg --fix` replaces the space-based indentation with tab-based indentation. The second run restores proper space-based indentation.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the following sample file:
```
entity sample is
  port(
    a, b  : in std_logic;
    d, ee : in std_logic
  );
end entity sample;
```
2. Run `vsg --fix -f sample.vhd`.
3. See that it is stuck at phase 5 with tab indentation.
4. Run `vsg --fix -f sample.vhd` again.
5. See that the file is successfully indented.

**Expected behavior**
* Space-based indentation in case of stop during run with `--fix`.
* One pass of `--fix` should clear all clearable issues.
* `entity_017` should suggest positive offsets

Have a good day!

jeremiah-c-leary, 2022-10-04:
Hey @Scafir,

I just pushed an update for this issue to the `issue-829` branch.  When you get a chance could you check it out on your end?

Thanks,

--Jeremy

Scafir, 2022-10-04:
Hi @jeremiah-c-leary ,
I tested it on my repo and it is running well! 
As always, many thanks for what you do :)

jeremiah-c-leary, 2022-10-04:
Sweet.  I will merge this to master.

--Jeremy

PR #830 (2022-10-04): Updates for issue #829
Rule port_026 was not copying tokens which caused rule entity_017 to misbehave.

  1)  Updated tests
  2)  Updated rule port_026

Resolves #829
closed 2022-10-04 by jeremiah-c-leary