jeremiah-c-leary/vhdl-style-guide #831: Junit result file does not get created if some vhdl files can not parsed

bae-pabr, 2022-10-04:
**Environment**
VHDL Style Guide (VSG) version: 3.12.2.dev5

**Describe the bug**
I want to run vsg on multiple files (file_list) and create a junit result file, but one of them is not parsable and no junit file is created.
I would like to skip unparsable files in the junit result file.

**To Reproduce**
see above

**Expected behavior**
Junit result file is created. 
It would be nice if the file name along with the message would be part of the result.
Alternative (if reporting not possible): I would like to skip unparsable files in the junit result file.

jeremiah-c-leary, 2022-10-05:
Hey @bae-pabr ,

I pushed an update for this to the `issue-831` branch.

The parsing error message is included in the JUnit XML file.

When you get a chance, could you check this out on your end?

Thanks,

--Jeremy

bae-pabr, 2022-10-05:
Hi Jeremy,

fix seems to work. (see shortend junit result below)

```
<?xml version="1.0" ?>
<testsuite errors="0" hostname="ATFKPCPABR" failures="17" timestamp="2022-10-05T02:51:21" tests="17" time="0" name="vhdl-style-guide">
  <properties>
  </properties>
[...]
  <testcase name="[...]" time="0" classname="failure">
    <failure type="Failure">
      
Error: Unexpected token detected while parsing element_declaration @ Line 40, Column 48 in file [...]
       Expecting : ;
       Found     : (

    </failure>
  </testcase>
 [...]
  <system-out>
  </system-out>
  <system-err>
  </system-err>
</testsuite>
```

Thank you for fixing this issue,
Patrick

jeremiah-c-leary, 2022-10-05:
Sweet!!

I will get this merged to master.

--Jeremy

PR #836 (2022-10-05): Updates for issue #831
Files which fail to parse will will be included in JUnit output.

  1)  Added test
  2)  Output parse failures to JUnit files

Resolves #831
closed 2022-10-05 by jeremiah-c-leary