jeremiah-c-leary/vhdl-style-guide #832: Generate statements do not end horizontal alignment blocks

Scafir, 2022-10-04:
**Environment**
CentOS Linux 7 (Core)
VHDL Style Guide (VSG) version: 3.12.2+zip.file

**Describe the bug**
Generate statements do not end horizontal alignment blocks, which results in constructs such as:
```
  a                                    <= b;
  gen : if '1' = '1' generate
    anExtraLoooooooooooooooooooongName <= c;
  end generate gen;
  aSlighltyLongerName                  <= d;
```

Note that this situation does not happen if the `blank_line` rule group is enabled (which is the default), and is worsened when using 4 spaces for tabulation.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the sample files:
`sample.vhd`
```
architecture behavioral of sample is
begin
  a                                    <= b;
  gen : if '1' = '1' generate
    anExtraLoooooooooooooooooooongName <= c;
  end generate gen;
  aSlighltyLongerName                  <= d;
end architecture behavioral;
```
`sampleRule.yaml`
```
rule:
  group:
    blank_line:
      disable: true
```
2. Run `vsg --fix -c sampleRule.yaml -f sample.vhd`
3. See the weird alignement between blocks.

**Expected behavior**
Generate statements should have the same impact as if statements in processes: end the alignment block.

jeremiah-c-leary, 2022-10-04:
Hey @Scafir,

I can see the desire to not align the `<=` within the generates to those outside it.

I will add this to my list.

--Jeremy

jeremiah-c-leary, 2022-10-04:
Hey @Scafir,

I pushed  an update for this issue to the `issue-832` branch.  I added the parameter `generate_statement_ends_group` to rule `concurrent_006`.  If you set this to `true` it should give the behavior you are looking for.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-10-07:
Hey @Scafir ,

I am planning on doing a release this weekend and was wondering if you had a chance to validate this is working for you.

Thanks,

--Jeremy

Scafir, 2022-10-10:
Hi @jeremiah-c-leary ,

My bad for the late message, I had a few things coming up last week and didn't check the messages. I hope it did not bother you too much.

I am happy with `generate_statement_ends_group` and how it works. I would advocate for having it as default, as it behaves quite a bit like if/else statements which is set by default.

Have a good day!

jeremiah-c-leary, 2022-10-10:
Morning @Scafir ,

> I am happy with generate_statement_ends_group and how it works.

Sweet!!

> I would advocate for having it as default, as it can behave quite a bit like if/else statements which is set by default.

I struggled with this one.  My first implementation had it on by default, but then I worried it would adversely affect people code base.  However, the more I think about it enabling `generate_statement_ends_group` makes sense.

I have pushed an update to enable it by default and will merge this to master.

Thanks,

--Jeremy

PR #835 (2022-10-10): Updates for issue #832
Added parameter to allow generate statement keywords to end groups.

  1)  Added tests
  2)  Added generate_statements_ends_group option to base rule configuring_keyword_alignment_rules
  3)  Updated documentation

Resolves #832
closed 2022-10-10 by jeremiah-c-leary