jeremiah-c-leary/vhdl-style-guide #833: Pragma vhdl_comp_on/off self destruct + are they truly respected?

Scafir, 2022-10-04:
**Environment**
CentOS Linux 7 (Core)
VHDL Style Guide (VSG) version: 3.12.2+zip.file

**Describe the bug**
`vsg` wants to inserts a white space in front of the --vhdl_comp_off or --vhdl_comp_on, which would effectively disable them. Moreover, some errors are still getting reported for the ignored lines (`whitespace_007`, in my case).

**To Reproduce**
Steps to reproduce the behavior:
1. Use the following `sample.vhd` file:
```
architecture behavioral of sample is
begin
    --vhdl_comp_off
    @my_autotool_variable@
    --vhdl_comp_on
end architecture behavioral;
```
2. Run `vsg -f sample.vhd`
3. See comment/whitespace errors
5. Run `vsg --fix -f sample.vhd` twice
6. See parsing error

**Expected behavior**
The pragma should not be modified, no error should be reported for anything between them.

jeremiah-c-leary, 2022-10-04:
Hey @Scafir,

I see the insertion of whitespace after the `--` and if I run VSG on the resulting file I get a parsing error:
```mono
Error: Unexpected token detected while parsing architecture_body @ Line 6, Column 6 in file None
       Expecting : end
       Found     : @my_autotool_variable@
```
Let me add a pragma exclusion to the `comment_100` rule and see if it addresses the `whitespace_007` issue also.

--Jeremy

Scafir, 2022-10-04:
Hi @jeremiah-c-leary ,
I just checked and it is running great (same behavior as the previous version of `vsg` on my whole codebase, except the file using pragmas. 
As you have noticed, I am writing a lot of bug reports. I really like your quick response, but I hope I am not wearing you down! I can always wait if you have other matters to attend to :) 
Thank you again for your excellent work!

jeremiah-c-leary, 2022-10-04:
Hey @Scafir,

> I just checked and it is running great (same behavior as the previous version of vsg on my whole codebase, except the file using pragmas.

A little confused by this sentence.  I think you are saying the fix did not affect files without the pragmas, but now the file with the  pragmas are working now, correct?

> As you have noticed, I am writing a lot of bug reports. I really like your quick response, but I hope I am not wearing you down! I can always wait if you have other matters to attend to :)

Keep them coming!  I only have a small code base to run on so I end up missing issues you are finding.

--jeremy

Scafir, 2022-10-05:
Hi,

Sorry if my answer was confusing: your fix is running as intended, and does not hinder the rest of the processing: all is good for merging on my end.

I will do my best to keep them coming!

jeremiah-c-leary, 2022-10-05:
Sweet!!

I will get this merged to master.

--Jeremy

closed 2022-10-05 by jeremiah-c-leary

Scafir, 2022-11-15:
Hi @jeremiah-c-leary ,

Unfortunately, this needs to be reopened, as I am getting whitespace_007 issues on the following sample file:
```
architecture rtl of fifo is

begin

    --vhdl_comp_off
    sdaIobufE : component IOBUF
        generic map (
            DRIVE      => 12,
            IOSTANDARD => "DEFAULT",
            SLEW       => "SLOW"
        )
        port map (
            O  => i2cSdaIn,
            IO => i2cSdaxDIO,
            I  => i2cSdaOut,
            T  => i2cSdaOut
        );
    --vhdl_comp_on

end architecture rtl;
```
Please tell me if you'd like me to open a new issue for this. 
In the meantime, I hope you are doing well!

jeremiah-c-leary, 2022-11-15:
Morning @Scafir ,

Could you open a new issue?

Thanks,

--Jeremy

jeremiah-c-leary, 2022-11-21:
Morning @Scafir ,

I created issue #878 and pushed a fix for this to master.

--Jeremy

PR #834 (2022-10-05): Updates for issue #833
Pragma comments were being update by rule comment_100 when they should not have been

  1)  Added tests
  2)  Updated classification of pragmas to the pragma itself

Resolves #833