jeremiah-c-leary/vhdl-style-guide #837: Rule group alignment: no_alignment is documented but does not exist

Scafir, 2022-10-07:
**Environment**
CentOS Linux 7 (Core)
VHDL Style Guide (VSG) version: 3.12.2+zip.file

**Describe the bug**
Even though `no_alignment` for the rule group `alignment` is documented, setting it does nothing, as demonstrated in the before/after in the documentation. Moreover, there is no occurrence of the keyword in the entire code-base (except in documentation). 
Is it possible to implement this feature?
At the same time, it may be good to issue an error message if a string in the rules is not matching any know keyword, thus simplifying the discovery of this kind of bug, as well as preventing head scratching due to typos.

I am interested in the `no_alignment` rule, because it would allow new commits to my code-base to not modify unrelated lines. I have had to deal (unrelated to `vsg`) with conflicts involving commits re-indenting ports containing dozens of signals at the same time as changing some of them, and it is a true nightmare. 

**To Reproduce**
Steps to reproduce the behavior:
1. Use:
    `sampleRules.yaml`
```
rule:
  group:
    alignment:
      no_alignment: true 
  ```
   `sample.vhd`
```
architecture behavior of sample is
  signal wr_en : std_logic;
  signal rd_en : std_logic;

  constant c_short_period : time;
  constant c_long_period  : time;
begin
end architecture behavior;
```
2. Run `vsg -c sampleRules.yaml -f sample.vhd --fix`
3. See that keywords are not all left-aligned

**Expected behavior**
Left align the keywords:
```
  signal wr_en : std_logic;
  signal rd_en : std_logic;

  constant c_short_period : time;
  constant c_long_period : time;
```

jeremiah-c-leary, 2022-10-07:
Morning @Scafir ,

> Even though no_alignment for the rule group alignment is documented, setting it does nothing, as demonstrated in the before/after in the documentation. Moreover, there is no occurrence of the keyword in the entire code-base (except in documentation).

I must have either intended to implement the option or ended up removing it and not updating the documentation.

> At the same time, it may be good to issue an error message if a string in the rules is not matching any know keyword, thus simplifying the discovery of this kind of bug, as well as preventing head scratching due to typos.

This is something that I have been meaning to include for awhile.

> I am interested in the no_alignment rule, because it would allow new commits to my code-base to not modify unrelated lines. I have had to deal (unrelated to vsg) with conflicts involving commits [...]

Merging sucks.   It is unclear to me how a `no_alignment` option would help.  Could you elaborate?

Thanks,

--Jeremy

Scafir, 2022-10-07:
Hi @jeremiah-c-leary , I hope everything is well

> Merging sucks. It is unclear to me how a no_alignment option would help. Could you elaborate?

I am not quite sure yet `no_alignment` is the way to go, because it reduces the readability a bit and may displease colleges with a long tradition of alignment  VHDL. It however has one big advantage.

Let's say for some external reason, I must change names on a port. In this instance,`voltageChangeRdyxDI` to `voltageChangeReadyxDI` and `windIntervalxDI` to `windIntervalNxDI`. 

### Using horizontal alignment

Say I have the following code:
<details>
  <summary>Original Code</summary>

```
entity sample is
    port (
        -- Inputs
        axDI                 : in  std_logic;
        bbbbbbbbxDI          : in  std_logic_vector(32 - 1 downto 0)
        windIntervalxDI      : in  std_logic;
        voltageChangeInvlxDI : in  std_logic_vector(voltageChangeRegLengthC - 1 downto 0);
        voltageChangeRdyxDI  : in std_logic;
        compA1xDI            : in  compAccurateFrontendT;
        controlA1xDI         : in controlAccurateFrontendT;

        -- [...]
    );
end entity sample;

```

</details>

Since `voltageChangeReadyxDI` is now the longest word in the block, all of the port block must be re indented.
<details>
  <summary>Commited Code</summary>

```
entity sample is
    port (
        -- Inputs
        axDI                  : in  std_logic;
        bbbbbbbbxDI           : in  std_logic_vector(32 - 1 downto 0)
        windIntervalNxDI      : in  std_logic;
        voltageChangeInvlxDI  : in  std_logic_vector(voltageChangeRegLengthC - 1 downto 0);
        voltageChangeReadyxDI : in  std_logic;
        compA1xDI             : in  compAccurateFrontendT;
        controlA1xDI          : in  controlAccurateFrontendT;

        -- [...]
    );
end entity sample;
```

</details>

Now if someone reviews the code, he will see the diff:
```
4,10c4,10
<         axDI                  : in  std_logic;
<         bbbbbbbbxDI           : in  std_logic_vector(32 - 1 downto 0)
<         windIntervalNxDI      : in  std_logic;
<         voltageChangeInvlxDI  : in  std_logic_vector(voltageChangeRegLengthC - 1 downto 0);
<         voltageChangeReadyxDI : in  std_logic;
<         compA1xDI             : in  compAccurateFrontendT;
<         controlA1xDI          : in  controlAccurateFrontendT;
---
>         axDI                 : in  std_logic;
>         bbbbbbbbxDI          : in  std_logic_vector(32 - 1 downto 0)
>         windIntervalxDI      : in  std_logic;
>         voltageChangeInvlxDI : in  std_logic_vector(voltageChangeRegLengthC - 1 downto 0);
>         voltageChangeRdyxDI  : in  std_logic;
>         compA1xDI            : in  compAccurateFrontendT;
>         controlA1xDI         : in  controlAccurateFrontendT;
```
Which is pretty unreadable.

### Using left alignement

The style of each line is independent which give an original code of:
 <details>
  <summary>Original Code</summary>

```
entity sample is
    port (
        -- Inputs
        axDI : in  std_logic;
        bbbbbbbbxDI : in  std_logic_vector(32 - 1 downto 0)
        windIntervalxDI : in  std_logic;
        voltageChangeInvlxDI : in  std_logic_vector(voltageChangeRegLengthC - 1 downto 0);
        voltageChangeRdyxDI : in  std_logic;
        compA1xDI : in  compAccurateFrontendT;
        controlA1xDI : in  controlAccurateFrontendT;

        -- [...]
    );
end entity sample;

```

</details>
Which admittedly does not look as good as the previous one, but with the help of a few newlines could be made better (in any case, doxygen comments usually messes up the readability most of the time).

The committed code would then be:
<details>
  <summary>Commited Code</summary>

```
entity sample is
    port (
        -- Inputs
        axDI : in  std_logic;
        bbbbbbbbxDI : in  std_logic_vector(32 - 1 downto 0)
        windIntervalNxDI : in  std_logic;
        voltageChangeInvlxDI : in  std_logic_vector(voltageChangeRegLengthC - 1 downto 0);
        voltageChangeReadyxDI : in  std_logic;
        compA1xDI : in  compAccurateFrontendT;
        controlA1xDI : in  controlAccurateFrontendT;

        -- [...]
    );
end entity sample;
```

</details>

Which give a diff of:
```
6c6
<         windIntervalNxDI : in  std_logic;
---
>         windIntervalxDI : in  std_logic;
8c8
<         voltageChangeReadyxDI : in  std_logic;
---
>         voltageChangeRdyxDI : in  std_logic;
```
Which is much better!

This is even more true in case of alignment of instantiated ports or functions with many variables. Most of the time, those are a continuous alignment block and if the longest port name is modified, all of the lines are touched.

jeremiah-c-leary, 2022-10-07:
Hey @Scafir ,

Thanks for the clarification.   At this moment there are two possible solutions:

  Option 1:  Disable the `compact_alignment` option in rule `entity_017`
  Option 2:  Disable rule `entity_017` and enforce a single space before the colon via rule `port_020`

## Option 1

You could put the colon at some "reasonable" distance to the right of the port identifiers:

```vhdl
         axDI                            : in  std_logic;
         bbbbbbbbxDI                     : in  std_logic_vector(32 - 1 downto 0)
         windIntervalNxDI                : in  std_logic;
         voltageChangeInvlxDI            : in  std_logic_vector(voltageChangeRegLengthC - 1 downto 0);
         voltageChangeReadyxDI           : in  std_logic;
         compA1xDI                       : in  compAccurateFrontendT;
         controlA1xDI                    : in  controlAccurateFrontendT;
```
If you disable the `compact_alignment` option:
```yaml
rule:
  entity_017:
    compact_alignment: False
```
then VSG will align the colon to the right most colon.

So as long as the new port does not push the colon past the existing colons, the existing ports would not change.

## Option 2

Rule `port_020` enforces the number of spaces before the colon in port declarations.  It is currently set that is must be greater than or equal to one.  This can be configured to enforce a single space.  However, the alignment rule `entity_017` would also have be disabled:
```yaml
rule:
  port_020:
    number_of_spaces : 1
  entity_017:
    disable : True
```

Would either of these options work for you?

--Jeremy

Scafir, 2022-10-10:
Hi @jeremiah-c-leary ,

Thanks for the precise answer. I do not think that either option go far enough for me, as they do not address the rest of keyword alignment rules (though I did not think of disabling the compact alignment, which is a pretty good workaround).

While looking for how different style guides were dealing with horizontal alignment, I found that google [does](https://google.github.io/styleguide/javaguide.html#s4.6.3-horizontal-alignment) [not](https://google.github.io/styleguide/jsguide.html#formatting-horizontal-alignment) like it much (it is discouraged, but not forbidden). I have not been able to find much more literature on it except blogs. 
After reading what I could find on the subject and giving it a good though, I think that good solution for me would be to leave the user responsible for alignment and thus disabling the horizontal alignment rules. It may be interesting to issue warnings when more than one white-space not part of indentation is used on any particular line, but the work for implementing that may not be quite worth it. 

The more I look into this, the more I find it hard to have a "one fits all" type of rule for horizontal alignment. It is really helpful in some situations:
```
        inputsOr(I) <= (not input1xDI and matrixCoefficientsxDI(I)(0)) or
                       (    input1xDI and matrixCoefficientsxDI(I)(1)) or
                       (not input2xDI and matrixCoefficientsxDI(I)(2)) or
                       (    input2xDI and matrixCoefficientsxDI(I)(3)) or
                       (not input3xDI and matrixCoefficientsxDI(I)(4)) or
                       (    input3xDI and matrixCoefficientsxDI(I)(5)) or
                       (not input4xDI and matrixCoefficientsxDI(I)(6)) or
                       (    input4xDI and matrixCoefficientsxDI(I)(7));
```
but in others may lead to have a "blast radius" around changes, which causes all kinds of different problems.

jeremiah-c-leary, 2022-10-10:
Morning @Scafir ,

> I found that google [does](https://google.github.io/styleguide/javaguide.html#s4.6.3-horizontal-alignment) [not](https://google.github.io/styleguide/jsguide.html#formatting-horizontal-alignment) like it much (it is discouraged, but not forbidden).

Thanks for the link.  The reasoning they provide is sound, however one must way the costs of merging against the benefits of aligning.  One could argue you should optimize for whatever action you do most.  If you read code more then maybe you use column alignment.  If you merge more then maybe you do not.

> but in others may lead to have a "blast radius" around changes, which causes all kinds of different problems.

Everything is a trade off and I think it really comes down to what causes the least amount of pain.  When the  amount of pain is variable and depends on the person doing the task, it makes it impossible to satisfy everybody.

> I think that good solution for me would be to leave the user responsible for alignment and thus disabling the horizontal alignment rules. It may be interesting to issue warnings when more than one [...]

What are you imaging is involved in the work?

Regards,

--Jeremy

Scafir, 2022-10-10:
There is for sure a tradeoff between readability of the code and playing nice with version control. Each project needs to decides what its priorities are. 

> What are you imaging is involved in the work?

Without knowing much about `vsg`'s codebase, I expect the work in itself to not be very complicated, as it can probably be done by:
1) removing all whitespaces up to first character
2) checking that there is not more than one consecutive whitespace

The biggest uncertainties for me are how to make that play nice with the rest of the rules, and how to make life easy on the user, as enforcing the line to be compact is asking for trouble. 

There will always some situation where some amount of alignment is desirable. This then means that the severity of the rule would have to be `warning`. In turn, I expect such a rule to generate a lot of noise in big code-bases, which is not great either, as past a certain numbers of warnings, even I would probably not bother to check if my changes generated a new one. 

Finally, I guess that for this use case, disabling the alignment checks would end up being the desirable solution, as this would preserve the output of `vsg` clear and provide flexibility to the developer and maintainer.

I am still a bit uncertain about this whole thing. I guess that I need time for all this to settle down and to be clear about what my needs are.

jeremiah-c-leary, 2022-10-10:
Hey @Scafir ,

> The biggest uncertainties for me are how to make that play nice with the rest of the rules, and how to make life easy on the user, as enforcing the line to be compact is asking for trouble.

One of the challenges with VSG, since it is configurable, is how to configure it.  Some concepts such as `global` and `rule_groups` help with configuring multiple rules.
```yaml
rule:
  global:
    indentSize : 2
  group:
    alignment:
      disable: true
    whitespace:
      number_of_spaces: 1
  concurrent_003:
    disable: False
```
> I am still a bit uncertain about this whole thing. I guess that I need time for all this to settle down and to be clear about what my needs are.

I will keep this open and ping you from time to time to see where you stand on this.  So a the moment I will assume there is no action on my part until I hear back from you.

Regards,

--Jeremy

jeremiah-c-leary, 2022-12-07:
Hey @Scafir ,

Just wanted to ping you on this and see where your thoughts are.

--Jeremy

Scafir, 2022-12-08:
Hey @jeremiah-c-leary,
I have not had the occasion to have a true discussion about that within my team. However, the more time passes, the more I am confident in the fact that I do not want to enforce horizontal alignment in every instance.
Writing this comment, I realize they are different categories for this alignment, each of which I feel different about.

### End of line alignment:
This is when items usually at the end of the line are aligned (e.g: comments). This bothers me because of readability reasons, as one long line may displace all other items by an unreasonable amount. I believe that these rules should be disabled in most "complex" usecases.

Concerned rules:
* `architecture_027`
* `entity_018`
* `entity_020`
* `process_035`

### Low value alignment:

This is the category where people may disagree. Rules I mention here add some esthetics to the code, but at the expense of version control. I guess that disagreement here will most likely stem from how the developer works. For example, I am a big fan of the "search" function (highlighting all occurrences of the word of interest). In this case, it is very clear where what I am looking for is and alignment does not help me much.
For someone else, not using this feature, alignment helps the eye to find the start of the word, resulting is some added value.

Concerned rules:
* `concurrent_400`: (we have huge functions where the "blast radius" breaks git reviews).
* `architecture_026`
* `concurrent_006`
*  `instantiation_010`
*  `signal_012`

### High value alignment:

I want to put one rule above the other, as in this case the alignment may help global understanding and may not be replaced by a search. I am talking about `entity_017`, which aligns the `in` and `out` keywords in entities. In this case, alignment makes it easier to understand what the entity does and how it works, as it makes finding the inputs and outputs easy. I still have it disabled, but now that I am writing this, I may actually change my mind.

### Conclusion

I would suggest re-evaluating whether the "End of line alignment" are worth having by default. They may work well in simple examples, but break when complexity of the code is high (e.g. having a signal `natural from 0 to 2048` will displace all surrounding comments from `architecture_027` or forcing either to have noisy line returns).

For "Low value alignment", the fight is open. I think having a grouped option to disable all of them at once would be nice to have. I am unsure whether `entity_017`, should be bundled with them. Is it possible to disable a rule group and re-enable a particular disabled rule manually?

Regards,
Clyde

jeremiah-c-leary, 2022-12-15:
Evening @Scafir ,

I believe I understand your position on the different priority levels of alignment.  However, I am wondering if dividing rules into different value buckets really solves the git review problem.  Any change in alignment could cause changes in other lines.  It really seems you can never get away from the blast radius effect.

Maybe this can be solved using git itself.  Git has an ignore whitespace option `--ignore-all-spaces`.  If you use this option then any alignment differences would be completely ignored.

Regards,

--Jeremy

Scafir, 2022-12-21:
Hi @jeremiah-c-leary,

>  However, I am wondering if dividing rules into different value buckets really solves the git review problem. It really seems you can never get away from the blast radius effect.

I believe that dividing the rules in different buckets can help, but not everyone will wish for the same one.. It probably would not be fruitful to implement them on your side.
In any case, I agree. As long as there is some horizontal alignment this problem is bound to happen.

Thanks for the git option! I never considered it existing, but it'll make my life much easier.

My conclusion to this whole story:
If one wants to not use horizontal alignment, then simply specify
```
rule:
  group:
    alignment:
      disable: True
```
Then, if you wish to enable some of the concerned rules, or change severity, do that with:
```
rule:
[...]
  myRule:
    disable: False
    severity: warning
```

The only thing that could be nice would be to have is a list of all the group's concerned rules in the documentation.

Regards,
Clyde

jeremiah-c-leary, 2022-12-21:
Morning @Scafir ,

> The only thing that could be nice would be to have is a list of all the group's concerned rules in the documentation.

I could add that to the documentation under the **Rule Groups**  section.

--Jeremy

Scafir, 2022-12-21:
Hi @jeremiah-c-leary ,
`grep`ing the documentation for alignement rules, I get the following concerned rules. I hope it will be useful for you:
```
after_002
architecture_026
architecture_027
architecture_029
architecture_400
assert_400
block_400
block_401
block_402
case_generate_statement_400
case_011
component_017
component_020
concurrent_003
concurrent_006
concurrent_008
concurrent_009
concurrent_400
concurrent_401
constant_012
constant_014
declarative_part_400
entity_017
entity_018
entity_020
function_012
function_015
generate_400
generate_401
generate_402
generate_403
generate_404
generate_405
if_009
instantiation_010
instantiation_029
library_009
package_body_400
package_body_401
package_body_402
package_019
package_400
package_401
package_402
procedure_call_400
procedure_call_401
procedure_010
procedure_401
procedure_410
procedure_411
procedure_412
process_020
process_028
process_031
process_033
process_034
process_035
process_400
process_401
report_statement_400
selected_assignment_400
sequential_004
sequential_400
sequential_401
sequential_402
signal_012
signal_400
subprogram_body_400
subprogram_body_401
type_400
variable_assignment_004
variable_assignment_400
variable_assignment_401
variable_400
```

I believe that this issue can be closed. One day, when I'm confident enough in the quality of my rules, I may submit a style that could be used as a starting point for people looking to be more relaxed on horizontal alignment.

Thank you for your patience,
Clyde

jeremiah-c-leary, 2022-12-21:
Morning @Scafir ,

I have added documentation which includes a link to each rule in a rule group.

I will close this issue.

Thanks,

--Jeremy

PR #903 (2022-12-21): Issue #837: Adding listing of rules in rule groups.
Resolves #837
closed 2022-12-21 by jeremiah-c-leary