jeremiah-c-leary/vhdl-style-guide #842: Prefix/suffix of loop labels

avelure, 2022-10-18:
**Is your feature request related to a problem? Please describe.**
I see rules for prefix/suffix of loop labels is missing. It would be nice to have.

**Describe the solution you'd like**
Something like for the other labels.
The label is optional for us, so I assume that it will only error if it finds a label and it doesn't conform with the prefix/suffix.

jeremiah-c-leary, 2022-10-18:
Morning @avelure ,

I can add the prefix and suffix rules for loop labels.

> The label is optional for us, so I assume that it will only error if it finds a label and it doesn't conform with the prefix/suffix.

You are correct.  If a label exist and the new rules are enabled, then a error will occur only if the loop label exists.

Regards,

--Jeremy

jeremiah-c-leary, 2022-10-18:
Morning @avelure ,

I pushed an update for this issue to the `issue-842` branch.  When you get a chance could you check it out on your end and let me know if it meets your requirements?

Thanks,

--Jeremy

avelure, 2022-10-18:
Hi,
this also works well, thanks!

jeremiah-c-leary, 2022-10-18:
Hey @avelure ,

Awesome, I will get this merged to master so it can be in the next release.

--Jeremy

PR #845 (2022-10-18): Updates for issue #842
Adding prefix/suffix rules for loop labels:

  1)  Added documentation
  2)  Added test cases
  3)  Added rules
      a)  loop_statement_600
      b)  loop_statement_601

Resolves #842
closed 2022-10-18 by jeremiah-c-leary