jeremiah-c-leary/vhdl-style-guide #843: Casing for type enumerated literals

avelure, 2022-10-18:
**Is your feature request related to a problem? Please describe.**
We have a rule that type enumerated literals should be capitalized, while most other things are lowercase.
It seems to be missing from the type rules.

**Describe the solution you'd like**
A new casing rule.

Violation
type state_machine is (idle, write, read, done);

Fix
type state_machine is (IDLE, WRITE, READ, DONE);

jeremiah-c-leary, 2022-10-18:
Morning @avelure ,

I pushed an update for this issue to the `issue-843` branch.  When you get a chance could you check it out on your end and let me know how it works.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-10-18:
Hey @avelure ,

I will get this merged to master.

--Jeremy

PR #844 (2022-10-18): Updates for issue #843
Adding case rule for enumerates types in type definitions.

  1)  Added documentation
  2)  Added test case
  3)  Added rule type_500

Resolves #843
closed 2022-10-18 by jeremiah-c-leary