jeremiah-c-leary/vhdl-style-guide #847: File globbing

avelure, 2022-10-18:
**Environment**
Windows, master.

**Describe the bug**
According to documentation the `-f` and `-c` parameters allow for file globbing. 
However, usign globbing only gives a file not found error.
The issue seems to be that `os.path.isfile` is called directly, maybe that works on Linux, instead `glob.glob` or `glob.iglob` should probably be used.

**To Reproduce**
Steps to reproduce the behavior:
`vsg -f *.vhd`
`VHDL Style Guide (VSG): error: argument -f/--filename: The file *.vhd does not exist.`

**Expected behavior**
It finds all the files in the folder.

jeremiah-c-leary, 2022-10-19:
Hey @avelure ,

I believe linux expands the glob before passing it to VSG.

I might be able to expand the glob in the `__is_valid_file` function.  I will have to play around with it.

--Jeremy

jeremiah-c-leary, 2022-10-20:
Hey @avelure ,

I pushed an update to this to the `issue-847`.  When you get a chance could you check it out and let me know how it goes?

Thanks,

--Jeremy

jeremiah-c-leary, 2022-11-07:
Morning @avelure ,

Just wanted to ping you to see if you had a change to validate the update.

Thanks,

--Jeremy

avelure, 2022-11-19:
Yes, thanks, this works for both files and configuration.

jeremiah-c-leary, 2022-11-19:
Morning @avelure ,

I will get this merged to master so it can be in the next release.

Thanks,

--Jeremy

PR #849 (2022-11-19): Updates for issue #847
Fixing globbing in the -f command line option.

Resolves #847
closed 2022-11-19 by jeremiah-c-leary