jeremiah-c-leary/vhdl-style-guide #850: alignment for "with select"

oletf, 2022-10-21:
Greetings,

I recently realized that the "with select" assignments don't get any alignment :
```vhdl
with mux select dout <=
          to_signed(+ 1, dout'length)when "11",
             to_signed(- 1, dout'length)when"10",
        to_signed(0, dout'length) when others;
```
stays as it is when running vsg (both alignment and spacing around the `when` don't change) .

Would it be possible to make a rule so that it becomes formatted as either :
```vhdl
with mux select dout <=
    to_signed(+ 1, dout'length) when "11",
    to_signed(- 1, dout'length) when "10",
    to_signed(0, dout'length) when others;
```
or
```vhdl
with mux select 
  dout <= to_signed(+ 1, dout'length) when "11",
          to_signed(- 1, dout'length) when "10",
          to_signed(0, dout'length) when others;
```
whichever is the easiest to implement for you (number of space isn't important)

also would it be possible to remove the space rule for unary +/- operators ?

Thanks forward!

jeremiah-c-leary, 2022-10-21:
Morning @oletf ,

Currently there are no rules for formatting with statements.  I do not use them, but if you are willing to work with me we can define the rules.

Here is an initial set of rules I can think of:

#### Indent rules
- [ ] indent of with keyword

#### Case rules
- [ ] case of with keyword
- [ ] case of select keyword
- [ ] case of when keyword

#### Whitespace rules
- [ ] single space after with keyword
- [ ] single space before select keyword
- [ ] single space before <=
- [ ] single space after <=
- [ ] single space before when keyword
- [ ] single space after when keyword

#### Structure rules
- [ ] no code after select keyword (configurable yes/no)
- [ ] no code after <= (configurable yes/no)
- [ ] no code after comma (configurable yes/no)
- [ ] when keyword on same line as waveform
- [ ] semicolon on same line as choice
- [ ] comma on same line as choice
- [ ] selected waveforms (waveform **when** choices) on a single line

#### Alignment rules
- [ ] alignment of target (configurable left,/select keyword?)
- [ ] alignment of waveforms (configurable left/<=)

Are there any possible rules I am missing or any of the above that do not seem valuable?

> also would it be possible to remove the space rule for unary +/- operators ?

Could you add another issue addressing this?  It will help me keep the two issues separate.

Thanks,

--Jeremy

oletf, 2022-10-24:
Hi,
I wouldn't need that much options personally, but other people definitely would want to tweak those so I think they're needed !
Couldn't think of another rule for now, looks all good to me.

Thanks a lot

jeremiah-c-leary, 2022-10-24:
Hey @oletf ,

Sounds good.  BTW, I am taking  a bit of break so I will not be getting to this soon.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-11-05:
Hey @oletf and @staerz ,

I pushed an initial update for this issue onto the `issue-850` branch and was wondering if you could take it for a spin.  Most of the new rules are mundane.

The interesting rule is `selected_assignment_001` which manages the structure of selected_assignments.  It has a corresponding [configuration page](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-850/docs/configuring_selected_assignment_structure_rules.rst) with a lot of options.

The alignment rule `selected_assignment_400` only supports left alignment currently.  It was going to limit it to this option unless someone requests a different method.

Let me know what you think.

Regards,

--Jeremy

oletf, 2022-11-08:
Hi,
I have to admit I can't find much to add to what @staerz said,
agreeing with all this.

jeremiah-c-leary, 2022-11-12:
Morning @staerz ,

> I see no problem with the defined rules in general.
>
> I do though think you're overdoing the job for the configuration of selected_assignment_001 a lot:

> A solution to that could be to actually split off this configuration into a dedicated rule such that the entire structure of with <input> select <signal> <= [<assignment> when <condition>][,|;] falls apart into multiple rules, each checking a dedicated part.

It is interesting you mention this, as `selected_assignment_001` was an experiment to see if I could combine multiple structure rules into a single rule.

If you look at the [block rules](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-850/docs/block_rules.rst),  the following rules deal with new lines in `block_statement`s:

* block_001
* block_003
* block_004
* block_005
* block_006

It also appears several rules are missing, one being the location of the semicolon.  One could also argue the optional items could  be included in a single structure rule also.

So I was wondering if it would be easier for the user to have one rule to configure the structure of a `selected_assignment` instead of multiple rules.  It seems easier to have them [documented](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-850/docs/configuring_selected_assignment_structure_rules.rst) in one place.  Although the documentation does not include an example above the configuration table.

> Looking further into the structure of it, I think the options `new_line_before_assignment`, `new_line_before_comma` and `new_line_before_semicolon` are nonsensical as it should be the standard (and there should be no discussion about it)

I agree.  I needed some way of communicating that new lines would be checked for those elements, and I get the add/remove/ignore for free.  I could change the options to only allow `remove`, but then it is not configurable.

So I think the question is which method, single rule or multiple rules, would be preferred.  Is it too much to ask for a single rule to cover so many individual items?  I can see the pros and cons for both but was wondering what you and @oletf think.

Regards,

--Jeremy

staerz, 2022-11-14:
> So I think the question is which method, single rule or multiple rules, would be preferred.

From a user perspective it doesn't matter too much as long as things are consistent.

From a developer perspective I would always try to solve identical issues with the same solution - i.e. apply the same (base) rule to the same structure.

In that sense, as a user I wouldn't mind sth. like the following:
- rule **a** determines how to format the `with x select y <=` section (i.e. 1 space only between the parts, upper/lower case of constituents - all that via dedicated config)
- rule **b** determines how to format the `when <condition>][,|;]` part (spaces, upper/lower case)
- rule **c** is an implied rule (set) which determines the actual assignment expression (simple, array, structure ...) according to the already defined assignment rules for signals (e.g. `signal_017`, `signal_400`)

So I think it's rather less about to put everything into one rule, but about not bugging a user to have to **redefine** the configuration of a new rule when an existing rule would do the trick (which the user has already configured to its preferences).
It's not that it's not too much to ask, but it's just very inconvenient.

jeremiah-c-leary, 2022-11-15:
Evening @staerz ,

> So I think it's rather less about to put everything into one rule, but about not bugging a user to have to redefine the configuration of a new rule when an existing rule would do the trick (which the user has already configured to its preferences).
It's not that it's not too much to ask, but it's just very inconvenient.

I think that is one of the guidelines I need to add to issue #866 where we can discuss the path forward.

So I think for this issue I will remove rule `selected_assignment_001`.  The rules I added address @oletf 's issue and I believe it also addresses your issue #853.   I will then addresses the structure of selected assignments as part of issue #866.

Regards,

--Jeremy

staerz, 2022-11-15:
Hi @jeremiah-c-leary,

I'm not sure to fully follow.

I just looked at the branch `issue-850` but it's currently an offspring from 3.13.0 and rebasing to master fails due to conflicts.
If you don't mind, please rebase that branch as I'm a bit confused of what I should expect: Which rules have been added and which not? (Just looking at the files being added/removed, I still see plenty of files named `select_assignment` which makes me wonder as you state you removed this rule again - shouldn't these files then be gone as well?)

jeremiah-c-leary, 2022-11-16:
Evening @staerz ,

I merge master up to branch `issue-850` and resolved the conflicts.

I added these [rules](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-850/docs/selected_assignment_rules.rst) and deprecated this [rule](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-850/docs/with_rules.rst).

--Jeremy

staerz, 2022-11-16:
Hi @jeremiah-c-leary,

OK, thanks for cleaning up.

Looking at the documentation, I'm wondering: For `selected_assignment_400` your example uses a variable (assignment operator is `:=`). Is that according to the LRM? I thought that only signals could be assigned with that construct.

Given what we discussed earlier, I would still appreciate a lot if the documentation would change the example to the "proper" styel, i.e. to have all until the assignment operator on one line which will automatically make the indentation of the actual assignments much more consistent with the 2 spaces per level.

If I take your example and extend it a bit to check for comments:
```vhdl
  -- here a comment
  with mux_sel select
    -- here another one
    addr <= give("0000" +
     -- some more
      "1111" -
     -- and more
      "0101") when 0,
    -- and another one
    "0001" when 1,
    -- and the last one
    "1111" when others;
```
Then I get complaints on `-- some more` and `-- and more`: VSG wants me to use 4 spaces here where it should actually be 6, so it seems the indentation for comments doesn't follow the indentation of the elements of this expression. (Note that I use the default for Multiline Indent Rules).

The very same is true when I format it this way:
```vhdl
  -- here a comment
  with mux_sel select addr <=
    -- here another one
    give("0000" +
     -- some more
      "1111" -
     -- and more
      "0101") when 0,
    -- and another one
    "0001" when 1,
    -- and the last one
    "1111" when others;
```

It appears that `selected_assignment_102` does actually not complain if there is not a single space but a line break instead. Is that the intended behaviour? Again, I would argue that there should be no line break in the entire construct until the assignment operator (and I just checked, I can basically add line breaks anywhere without complaints ... I want complaints :rofl: )

jeremiah-c-leary, 2022-11-16:
Afternoon @staerz ,

> Looking at the documentation, I'm wondering: For selected_assignment_400 your example uses a variable (assignment operator is :=). Is that according to the LRM? I thought that only signals could be assigned with that construct.

There are four possible selected assignments in the LRM:
```text
concurrent_selected_signal_assignment ::=
    with expression select [ ? ]
        target <= [ guarded ] [ delay_mechanism ] selected_waveforms ;

selected_force_assignment ::=
    with expression select [ ? ]
        target <= force [ force_mode ] selected_expressions ;

selected_variable_assignment ::=
    with expression select [ ? ]
        target := selected_expressions ;

selected_waveform_assignment ::=
    with expression select [ ? ]
        target <= [ delay_mechanism ] selected_waveforms ;
```
with one of them being for variables.  Which makes sense as variables are similar to signals.

> Given what we discussed earlier, I would still appreciate a lot if the documentation would change the example to the "proper" styel, i.e. to have all until the assignment operator on one line which [...]

Oh yeah...I can do that.  I will also change the `:=` to `<=`.

> If I take your example and extend it a bit to check for comments:

The indenting is increased by 1 when the `with` keyword is hit and then decremented by 1 when the semicolon is hit.  I believe the multiline indent code ignores comments and leaves it up to comment indent rule to handle it.  Although thinking about it, that is probably not preferred as one could argue we are aligning at this point and not indenting.

I assume you would want the comments aligned like this:
```vhd
  -- here a comment
  with mux_sel select addr <=
    -- here another one
    give("0000" +
      -- some more
      "1111" -
      -- and more
      "0101") when 0,
    -- and another one
    "0001" when 1,
    -- and the last one
    "1111" when others;
```
Where `-- here a comment` would be handled by rule `comment_010` and the other comments would be handled by `selected_assignment_400`.

I do not think that will be too hard to implement.

> It appears that selected_assignment_102 does actually not complain if there is not a single space but a line break instead. Is that the intended behaviour? 

Yeah, that is the intended behavior.  If `select` at the end of a line, either followed by a carriage return, comment, whitespace and then a comment the rule does not apply.

The rule `selected_assingment_100` would be responsible for removing new lines and forcing `target` to be on the same line as the `select` keyword.

Regards,

--Jeremy

staerz, 2022-11-16:
Hi @jeremiah-c-leary,

> Oh yeah...I can do that. I will also change the := to <=.

OK, that would be great. And OK, if this construct also works for variables, then I just learned something, thank you! (Just wanted to double check!).

> I assume you would want the comments aligned like this:

Yes, exactly like this. Note that the indent for `-- some more` and `-- and more` would depend on the configuration of `selected_assignment_400`: If `align_paren` is `yes`, then also the comment should move (but I use the `no` option).

> The rule selected_assingment_100 would be responsible for removing new lines and forcing target to be on the same line as the select keyword.

"Would" like in "would if it were implemented" or like "it actually will if you configure it accordingly"? I wouldn't know how I could currently achieve that - and yes, I want to force it on one line!

jeremiah-c-leary, 2022-11-16:
Hey @staerz ,

> "Would" like in "would if it were implemented" or like "it actually will if you configure it accordingly"? I wouldn't know how I could currently achieve that - and yes, I want to force it on one line!

Would as in the original `selected_assignment_100` rule was able to enforce new lines before and after tokens before I "removed" it.  I can add some structure rules to enforce the structure of `selected_assignments`.  

One thought I had was to have a single rule to remove newlines before all semicolons and commas instead of having a rule per production.  This would help reduce the number of rules.

--Jeremy

staerz, 2022-11-16:
Hi @jeremiah-c-leary,

> I can add some structure rules to enforce the structure of selected_assignments.

Yes, please, if possible. Again, I cannot see how anybody ever would want to split this over multiple lines and I think one rule to rule them all would be enough, i.e. one rule to check if that is all in one line and at the same time that things are separated by exactly one space (or configurable if you want, but default = 1 space).
If splitting this in 2 rules (or more) is the way to go, then fine, but I think less rules are better.

> One thought I had was to have a single rule to remove newlines before all semicolons and commas instead of having a rule per production. This would help reduce the number of rules.

100 times :100:! Absolutely what you should do.

In the same spirit you can also have 1 rule to check that there's only 1 semicolon (i.e. 1 "instruction") per line. Anything else would be a good coding style violation, no matter in which context it appears. (For commas, I think we agree that it's a different story as we easily end up in structures where line breaks may be intended and stylistically advisable.)

jeremiah-c-leary, 2022-11-18:
Morning @staerz ,

I have pushed an update that includes structure rules.  The rules will take the following code snippet:
```vhdl
  5   with
  6   expression
  7   select
  8   target
  9   <=
 10   "0000"
 11   when
 12   0
 13   , "0001"
 14   when
 15   1
 16   ;
```
and format it as this:
```vhdl
  5   with expression select target <=
  6     "0000" when 0,
  7     "0001" when 1;
```

I created 12 rules to handle this, but still not sure if atomizing the structure into 12 separate rules for having a single rule with 12 options is better.

Let me know what you think of the update.

I will start working on handling the indenting of comments.

Regards,

--Jeremy

staerz, 2022-11-21:
Hi @jeremiah-c-leary,

thanks for looking into this.

I do agree with you that we don't necessarily need 12 individual rules (visible to the user) for what we want to achieve here. It does though fall apart into 12 details (let's assume the count to be correct here).

If you look for grouping things, then I'd suggest to combine `selected_assignment_001` to `selected_assignment_004` into a single rule.
On the other hand, if the rule is violated, it is more explicit to the user where exactly the error is. I think at that point it's becoming a matter of taste.

Similarly rules `selected_assignment_005` to `selected_assignment_009` all go into the same direction ... but from a user perspective I think I'd prefer a detailed complaint on what exactly went wrong rather than a "There's code after 'transport', or 'force', ..." error message.

So in general I agree to all the rules 1 to 12, they define the structure and there is no configuration needed.

For `selected_assignment_010` I do see the potential that this might become a bit more complex as also structural assignments could be given here - would we be prepared for that? I.e. imagine a multiline array structure in the expression, I would expect [the according rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_array_multiline_structure_rules.html) to apply - but where goes the `when`? I think it should simply go in the same line as the last closing parenthesis.

Rule 11: In the doc, make it "This rule checks the choice is on the same line as the `when` keyword" (or "the according choice.").

I have taken a look into my code base and even considering possible edge cases for 10 and 11 I didn't find any where foreseeing a rule would be worth the effort over locally excluding the rule, so I think we're good.

So that's what I think of it just by reading the doc.

jeremiah-c-leary, 2022-11-21:
Morning @staerz ,

> If you look for grouping things, then I'd suggest to combine selected_assignment_001 to selected_assignment_004 into a single rule.
> On the other hand, if the rule is violated, it is more explicit to the user where exactly the error is. I think at that point it's becoming a matter of taste.

I agree that it is a matter of taste.  At times it makes sense to have a single rule with multiple options similar to the `constraints` structure rule.

> For selected_assignment_010 I do see the potential that this might become a bit more complex as also structural assignments could be given here - would we be prepared for that? I.e. imagine a [...]

Correct, then `when` is moved left to the next non-whitespace character.

> Rule 11: In the doc, make it "This rule checks the choice is on the same line as the when keyword" (or "the according choice.").

I can do that.

Also, I looked into the indenting of comments.  In order to handle aligning comments like this:
```vhdl
  -- here a comment
  with mux_sel select addr <=
    -- here another one
    give("0000" +
      -- some more
      "1111" -
      -- and more
      "0101") when 0,
    -- and another one
    "0001" when 1,
    -- and the last one
    "1111" when others;
```
I will need to remove the comments after the `with` keyword from the rule `comment_010`.  I believe this will require a change to my data structure and classifying process.  These efforts were originally slated for new year.

Do you require the indenting to be fixed in the above code before you would consider issue #853 addressed?

Would you consider this issue resolved?

Regards,

--Jeremy

staerz, 2022-11-21:
Hi @jeremiah-c-leary,

I'd have to actually run VSG with the implementation related to this issue in order to evaluate if it's addressed or not.

But yes, I agree to the general idea that VSG rule checks on the (generally speaking) structure of the `with x select y` construct being addressed solely by this issue, whereas #853 would take care of the comments in these.

Yes, for #853 I would though expect the comment indent checking to be correct (i.e. in line) with what ever structural solution we come up with (in this issue here).

But do not worry, there's no pressure. It is as always that I report issue as soon as I encounter them to have them off my table. How long they sit on yours is up to you and only you. (And again, this is not meant to create any pressure on your side.)

So then I'll reserve some time tomorrow to test the implementation of this issue and come back to you.

staerz, 2022-12-13:
Hi @jeremiah-c-leary,

thanks for your patience. I finally made it to test your latest implementation and do confirm that now the `with ... select` statement is properly handled by VSG, including the indentation of comments.

We might in future stumble over some peculiarities if the expressions get large or data types complex, but from my experience, these `with ... select` statements are usually in simple constructs and hence, for now, I'd say we're totally good.

I do confirm that #853 is also solved.

So from my side: Please go ahead and merge this! :+1:

jeremiah-c-leary, 2022-12-14:
Hey @staerz ,

Awesome, I will get this merged to master.

--Jeremy

PR #861 (2022-12-14): Issue 850
Resolves #850
Resolves #853
closed 2022-12-14 by jeremiah-c-leary