jeremiah-c-leary/vhdl-style-guide #851: Undocumented configuration features

staerz, 2022-11-01:
**Environment**
Ubuntu, VSG 3.13.0

**Describe the bug**
It's not a bug, it's a documentation flaw:

The rule `procedure_call_400` does allow configuration according to [Configuring Multiline Indent Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_multiline_indent_rules.html) but this is not documented.

By simply trial and error I found out that
```json
    "procedure_call_400": {
      "align_left": "yes",
      "align_paren": "no",
    },
```
for example is a valid configuration for it.

**To Reproduce**
Steps to reproduce the behavior:
1. RTMF

**Expected behavior**
Document the feature
- Note in `procedure_call_400` itself
- Note in list of Configuring Multiline Indent Rules

**Additional context**
Apparently these procedure call rules where introduced without me reviewing its documentation :rofl:

(And I'm wondering in how far these rules are different from port and generic map rules. The structure is identical and I would hope that under the hood the same base routines are used.)

jeremiah-c-leary, 2022-11-04:
Morning @staerz ,

Well...I decided to add tests for my documentation to validate links.  Since base rules, like the multi-alignment, map to a specific configuration document then all rules which extend the base rule will need to link to that document.  The documentation for the rules are contained in the docstrings for that rule.  So I added an attribute to my base rule class so I can set the link to the required documentation.

I have one test which checks if the attribute is set then the link must exist in the docstring.  When I ran my test for the multi-alignment rules I got the following:

[image]

So I was missing both `procedure_call_400` and `if_009`.

I already had a test that would validate the rule doc string matches the files used for generating the documentation on read the docs.

Then  I added a test to validate the rule links in the configuration documentation were correct.  The process is a little convoluted, but the result was a test with the following results:

[image]

So I had some rules configuration which did not have their attribute set:

* `concurrent_401`
* `constant_012`
* `sequential_402`
* `variable_assignment_401`

and I was missing several links:

* `if_009`
* `procedure_call_400`
* `signal_400`
* `variable_400`
* `variable_assignment_400`

I still need to expand this to cover all of the configurations, but now I have a set of tests to validate linking in the documentation.

I pushed the updated documentation to the issue-851 branch.  If you check it when you get chance I would appreciate it.

Thanks,

--Jeremy

staerz, 2022-11-04:
Hi @jeremiah-c-leary,

yes, I now see the link to the configuration in there :+1:

It's very good that you came up with a semi-automatic documentation consistency check!

jeremiah-c-leary, 2022-11-04:
Hey @staerz ,

Awesome, I will merge this to master.

--Jeremy

PR #860 (2022-11-04): Updates for issue #851
Adding tests to validate links to configuration documenation.

Resolves #851
closed 2022-11-04 by jeremiah-c-leary