jeremiah-c-leary/vhdl-style-guide #852: VSG cannot parse signal declaration with `'subtype`

staerz, 2022-11-01:
**Environment**
Ubuntu, VSG 3.13.0

**Describe the bug**
A signal declaration using the `'subtype` attribute makes VSG's parser fail and error out.

**To Reproduce**
Steps to reproduce the behavior:
1. Define a signal
2. Define another signal, using the `'subtype` attribute referencing the first signal
3. Run VSG over that code
4. See error

**Expected behavior**
I expect VSG to support this default VHDL syntax.

**Example Code**
The following example code will make VSG fail:
```vhdl
    --! Signal of explicit type definition
    signal foo : unsigned(31 downto 0);
    --! Signal of implicit type definition using attribute 'subtype
    signal bar : foo'subtype;
```
Produced output by VSG:
```
Error: Unexpected token detected while parsing signal_declaration @ Line 123, Column 45 in file foo_bar.vhd
       Expecting : ;
       Found     : subtype
```

**Additional context**
The similar behaviour would possibly be observed on constant or variable declarations.

jeremiah-c-leary, 2022-11-04:
Morning @staerz ,

I added support for attributes in subtype declarations and pushed an update to the issue-852 branch.  When you get a chance could you check it out on your end and let me know if it works on your end.

Thanks,

--Jeremy

staerz, 2022-11-04:
Hi @jeremiah-c-leary,

I'm amazed by your support!

Yes, I checked out the branch, installed it and tried it out, the failure is gone now (and the actual style issues of the previously failing file now show up as it should be).

Bazinga, or as you like it: Wooot wooot!

Thanks very much for this super quick intervention and bug fix!

(And just to inform you why "suddenly" now I'd report another set of issues: We have now entered the phase of our project to adapt the latest VSG release to a broader part of our code base, while so far I've only be testing on my part of it, so, I hope you can understand that it's only natural that edge cases would pop up.)

PR #856 (2022-11-04): Issue 852
Resolves #852
closed 2022-11-04 by jeremiah-c-leary