jeremiah-c-leary/vhdl-style-guide #853: Fix wrong level of indent for comments in `with x select y <=` constructs

staerz, 2022-11-01:
**Environment**
Ubuntu, VSG 3.13.0

**Describe the bug**
The indent level for comments in the `with x select y <=` construct is wrong (1 level too few).

**To Reproduce**
Steps to reproduce the behavior:
1. Create a `with x select y <=` construct with at least 2 cases, spread over 2 lines
2. Comment each of the assignments
3. Run VSG over it
4. See error

**Expected behavior**
The comment should be at the same level as the code of the particular `when` case.

Example code:
```vhdl
  with foo select bar <=
    -- upper byte of baz
    baz(15 downto 8) when 1,
    -- lower byte of baz
    baz(7 downto 0) when 2,
    -- half of upper byte and
    baz(15 downto 12) &
    -- half of lower byte of baz
    baz(3 downto 0)
      when 3,
    -- don't care
    (others => '-') when others;
```
is fixed to
```vhdl
  with foo select bar <=
  -- upper byte of baz
    baz(15 downto 8) when 1,
  -- lower byte of baz
    baz(7 downto 0) when 2,
  -- half of upper byte and
    baz(15 downto 12) &
  -- half of lower byte of baz
    baz(3 downto 0)
      when 3,
  -- don't care
    (others => '-') when others;
```

**Additional context**
Not sure what would happens with comments if they were in the line before the `when` keyword.

staerz, 2022-11-04:
Hi @jeremiah-c-leary,

I just looked up the current documentation and in deed, there are no rules for the `with x select y` VHDL construct (which Doulos catalogues under "Select").

Although Doulos is a great and reliable source for VHDL syntax, in this particular case I'd not go along their style in the example that they have.
Yes, from a perspective of explaining the construct, it is totally agreeable to put the signal to be assigned on the next line, but not from a structural perspective as such a statement can only assign one single signal.

For me, hence the following is (very) bad style:
```vhdl
with a select
  some_super_fancy_long_name_for_a_signal <= b when "HELLO",
                                             c when "hello",
                                             d when "world",
                                             '-' when others;
```
and instead the following, consistent with the rest, style should be applied:
```vhdl
with a select some_super_fancy_long_name_for_a_signal <=
  b when "HELLO",
  c when "hello",
  d when "world",
  '-' when others;
```

I hope you see where this goes (and you recognise the pattern from earlier discussions we had): 2 spaces indent where indent is due.
Note that this scales with what ever complex structures `b`, `c` and `d` would ever take (could be arrays of records ... think about the formatting here: I would EXPECT VSG to handle that properly and in line with all the formatting rules that we have previously carved out precisely and tediously).

In that sense, the `with a select` becomes a prefix of a usual signal assignment (for which rules exist) with a post `when y[,|;]` expression.

Now not sure how you integrate that into your existing code, but my big favour in your interest would be to reuse existing patterns.

Another note: There is currently one "[with rule](https://vhdl-style-guide.readthedocs.io/en/latest/with_rules.html?highlight=select)". In the spirit of #850, I think you should opt for one or the other, but not both.
I'd go with "with rules" for one simple reason: Only the `with a select [b <=]` construct is new to VSG, so the following rules (that you already foresee) would kick in:

1) indent of `with`
1) space between `with` and "condition"
1) space between "condition" and `select`
1) space between `select` and signal (which could potentially already be covered by any of the signal assignment rules), where one option in the context of the `with` construct could be to force a line break or not - but I wouldn't go for it as it introduces another unnecessary layer of indentation (that impacts any following lines...)

Anything beyond that should already be covered by existing signal assignment rules.

A lot of text for a tiny thing ... :rofl: but I hope you see my point.

jeremiah-c-leary, 2022-11-05:
Hey @staerz ,

Since this issue is related to issue #850 could we move the discussion to that issue?  Then we could close this issue when #850 is resolved.

Thanks,

--Jeremy

staerz, 2022-11-07:
Hi @jeremiah-c-leary,

sure. I looked for similar issues before opening this one but have no explanation why I wouldn't have seen #850.

PR #861 (2022-12-14): Issue 850
Resolves #850
Resolves #853
closed 2022-12-14 by jeremiah-c-leary