jeremiah-c-leary/vhdl-style-guide #854: Error assigning aggregate to std_logic_vector signal

amb5l, 2022-11-02:
**Environment**
I am running version 3.13 installed via `pip` into Python 3.10.8 on Windows 11 today.

**Describe the bug**
VSG reports rule violations correctly, but crashes when `--fix` is added to the command line, for the code below (minimised from original source). I suspect an issue with aggregates.

**To Reproduce**
Run VSG with and without `--fix` on the following VHDL source:
```
library ieee;
use ieee.std_logic_1164.all;

entity test_entity is
  generic (
    gen1     : bit     := '0';
    gen2     : bit     := '0'
  );
  port (
    d1        : in  std_ulogic;
    d2        : in  std_ulogic
  );
end entity test_entity;

architecture arch of test_entity is
  signal d       : std_logic_vector(1 downto 0);
begin
    process(rst,clki,clkdivi)
    begin
      if rising_edge(clkdivi) then
        d <=
          (d2 xor to_stdulogic(gen2)) &
          (d1 xor to_stdulogic(gen1));
      end if;
    end process;
end architecture arch;
```
Here is a transcript of the error:
```
C:\work>vsg -f bug.vhd --fix
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "D:\devtools\msys64\mingw64\lib\python3.10\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "D:\devtools\msys64\mingw64\lib\python3.10\site-packages\vsg\apply_rules.py", line 103, in apply_rules
    oRules.fix(
  File "D:\devtools\msys64\mingw64\lib\python3.10\site-packages\vsg\rule_list.py", line 158, in fix
    oRule.fix(self.oVhdlFile, dFixOnly)
  File "D:\devtools\msys64\mingw64\lib\python3.10\site-packages\vsg\rule.py", line 103, in fix
    self.analyze(oFile)
  File "D:\devtools\msys64\mingw64\lib\python3.10\site-packages\vsg\rule.py", line 140, in analyze
    self._analyze(lToi)
  File "D:\devtools\msys64\mingw64\lib\python3.10\site-packages\vsg\rules\multiline_array_alignment.py", line 66, in _analyze
    self.calculate_expected_indents(oToi, oLines)
  File "D:\devtools\msys64\mingw64\lib\python3.10\site-packages\vsg\rules\multiline_array_alignment.py", line 84, in calculate_expected_indents
    dExpectedIndent = self.analyze_align_left_false_align_paren_true(oToi, oLines)
  File "D:\devtools\msys64\mingw64\lib\python3.10\site-packages\vsg\rules\multiline_array_alignment.py", line 110, in analyze_align_left_false_align_paren_true
    check_middle_line(oLine, oLines, self.indentSize)
  File "D:\devtools\msys64\mingw64\lib\python3.10\site-packages\vsg\rules\multiline_array_alignment.py", line 160, in check_middle_line
    oLines.iNextIndent = oLines.lParens[-1].iExpectedColumn
IndexError: list index out of range
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "D:\devtools\msys64\mingw64\bin\vsg-script.py", line 33, in <module>
    sys.exit(load_entry_point('vsg==3.13.0', 'console_scripts', 'vsg')())
  File "D:\devtools\msys64\mingw64\lib\python3.10\site-packages\vsg\__main__.py", line 140, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "D:\devtools\msys64\mingw64\lib\python3.10\multiprocessing\pool.py", line 873, in next
    raise value
IndexError: list index out of range
```

jeremiah-c-leary, 2022-11-02:
Evening @amb5l,

I was able to replicate the error using your example.  I believe I know what is causing the issue.

--Jeremy

amb5l, 2022-11-03:
Thank you for jumping on this so quickly, and for creating VSG!

jeremiah-c-leary, 2022-11-04:
Morning @amb5l ,

> Thank you for jumping on this so quickly, and for creating VSG!

Thanks for the kind words.

>  I guess VSG is processing the RHS as if it is an array.

You are correct.  The array detection logic was not robust enough.

I pushed an update for this to the issue-854 branch.  When you get a chance could you check it out on your end and let me know if it works for you.

Thanks,

--Jeremy

amb5l, 2022-11-07:
That fixes this issue for me, thanks.

jeremiah-c-leary, 2022-11-07:
Morning @amb5l,

Awesome, I will merge this to master and then release 3.13.1.

--Jeremy

PR #857 (2022-11-07): Issue 854
Resolves #854
closed 2022-11-07 by jeremiah-c-leary