jeremiah-c-leary/vhdl-style-guide #863: Make VSG exit on encountering unknown rule

staerz, 2022-11-08:
**Environment**
Ubuntu, VSG 3.13.0 vs. older versions

**Describe the bug**
When switching between VSG versions usually also the configuration needs to change as new rules are added or old rules are deprecated which has an impact on the valid configuration options.

When VSG hits an unknown configuration, it loops unnecessarily repeating the same ERROR over and over.

**To Reproduce**
Steps to reproduce the behavior:
1. Have VSG rules defined in a file
2. Add an unsupported imaginary rule with some imaginary configuration (or a configuration of a deprecated rule)
3. run VSG
4. See error

Example:
```
$ vsg -c .vsg/rules/config.json .vsg/rules/disabled.json .vsg/rules/alignment.json .vsg/rules/casing.json .vsg_local/configuration.json -lr .vsg/rules
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
ERROR: Rule architectured_010 referenced in configuration could not be found
```

In this example, `.vsg/rules/config.json` contains an imaginary rule:
```json
    "architectured_010":{
      "action":"add",
    },
```

**Expected behavior**
Once reading an invalid configuration, VSG should simply exit with giving that error message exactly once (and maybe some context).

**Additional context**
Gets me every time I switch between versions that have new configs ... :wink:

jeremiah-c-leary, 2022-11-11:
Morning @staerz ,

The error is being repeated for every file being processed.  I added an escape when a configuration error occurs which should stop processing all files when a configuration error occurs.  I pushed an update to the `issue_830` branch.  When you get a chance could you check it out on your end?

Thanks,

--Jeremy

staerz, 2022-11-11:
Ah, OK, that explains why sometimes the list was longer and sometimes shorter.

Yes, I confirm that the update did what it's intended to do:
```bash
[staerz@staerz-dellp7550 LASP (master)]$ vsg $(cat .vsg/params.txt) | grep "| Error" -A10 -B10
ERROR: Rule concurrent_119 referenced in configuration could not be found
[staerz@staerz-dellp7550 LASP (master)]$ echo $?
1
```

That's much better!

(And yes, I figured out that you meant `issue_863` as a branch name. Maybe you allow a comment on your committing style: Yes, I appreciate that you actually put extensive commit messages, but for an easier quick grasp of the situation it would be advantageous if the first commit text (the first line, i.e. the text shown by `git log --oneline`) would be more significant. You have the habit of stating "Updates for issue #xyz". Yes, it is super good to have a reference to the issue, but as a user I'd be more interested what's behind this number. Yes, it is a matter of convention. You might consider the following: Just start the commit text with the hashtag and the issue number and then a brief summary. If that's not enough, then in subsequent lines you can add details. In our work flow we omit the reference to the issue in the single commits completely as we get that via the merge commits (it nicely relates issue and merge request numbers), but in a fast-forward git workflow like you apply it, I totally see why each commit message should reference the issue. So to give you a concrete example, I'd possibly messaged this one: "#863: Only spit error on faulty configuration once". Again, just a suggestions, it's your repo in the end.)

jeremiah-c-leary, 2022-11-12:
Morning @staerz ,

> And yes, I figured out that you meant issue_863 as a branch name

Hmm...not sure where my mind was, but it does illustrate how being consistent helps.

> You might consider the following: Just start the commit text with the hashtag and the issue number and then a brief summary.
> ...
> I'd possibly messaged this one: "https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/863: Only spit error on faulty configuration once"

The use of "Updates for issue..." is an artifact of my development before I started using feature branches.  When I started using feature branches I would still use "Updates for issue..." with every commit on the branch.  However, when I use squash merge to get the feature branch onto master I would have multiple "Updates for issue..." in the commit message.   So then I started only using "Updates for issue..." on the first commit so the squash commit message looked cleaner.

Your suggestion makes sense when using the `git log --oneline` as only seeing `Updates for issue #...` would not really mean anything.  I can also see synergy with my release notes:

```text
Bug Fixes
---------

  Issue #851:
    Updated links in documentation from rules to configuration pages.

  Issue #852:
    Parser would crash when attributes were used on subtype indications.

  Issue #854:
    Updateding array detection on signal assignments.

  Issue #855:
    Updating case rules to ignore names enclosed in double quotes.

```
Those single line descriptions are probably the level of information you would be looking for.  (Apparently I can't keep by tense consistent).

Hmmm...I wonder if I could automatically generate my release notes if I change the commit message to your suggestion.  If I could get the change log for every commit since the last release tag, I could parse the first line and format it like my release notes.  It might not get me 100% there, but it would save me time in generating the release notes.

I can see only upside to implementing your commit log suggestion, so will start using it on future issues.

I will merge this issue to master.

Thanks again,

--Jeremy

PR #865 (2022-11-12): Updates for issue #863
Invalid configurations will stop processing of all files.

Resolves #863
closed 2022-11-12 by jeremiah-c-leary