jeremiah-c-leary/vhdl-style-guide #864: All alignment rule for `:=` in signal, variable and constant defintion

staerz, 2022-11-10:
**Is your feature request related to a problem? Please describe.**
The feature request is related to style consistency.

**Describe the solution you'd like**
The rule [`entity_018`](https://vhdl-style-guide.readthedocs.io/en/latest/entity_rules.html#entity-018) makes sure that the `:=` default value operator is aligned for each generic and port in the entity declaration.

I would expect this to also be checked for out-of-port signal (, variable and constant) declarations (each port in an entity is a signal (and the `signal` keyword in that context is actually optional), each generic is a constant (and also the optional keyword)).

For constants there is [`constant_010`](https://vhdl-style-guide.readthedocs.io/en/latest/constant_rules.html#constant-010) which seems to go into that direction.
I do though doubt the effectiveness of it's `number_of_spaces` configuration as I cannot see the configuration be taken into account when dumping it using the vsg option `-oc`: What ever I do, the reported configuration stays:
```json
    "constant_010": {
      "disable": false,
      "fixable": true,
      "indentSize": 2,
      "indentStyle": "spaces",
      "number_of_spaces": ">=1",
      "phase": 2,
      "severity": "Error"
    },
```

I would expect
```json
    "constant_010":{
      "number_of_spaces": 1,
    },
```
to enforce 1 space only.

Note that the rule `constant_010` would actually have a potential to conflict with any rule that attempts to align the `:=` in case the configuration was enforced to anything else than `"number_of_spaces": ">=1"` and hence I'd suggest to abolish that rule in favour of the alignment rule (which should entail the check for `"number_of_spaces"` and take it into account).

**Describe alternatives you've considered**
To not care about it.

**Additional context**
This could fall into the aspect of reworking rules in a more global context as the principle would not only apply to constants, but also to signals (where default values should be avoided, but sometimes are OK) and variables.

Also, the check should not only be applied to the `entity` context but to any context in which a signal can be defined.

staerz, 2022-11-10:
PS: The configuration of `number_of_spaces` seems also not working for:
- `generic_014`
- `port_20`
- `sequential_003`
- `signal_006`
- `variable_006`
- `variable_assignment_003`
- `whitespace_007`

(I tried any rule that has `"number_of_spaces": ">=1"` as configuration, I didn't try others)

jeremiah-c-leary, 2022-11-12:
Morning @staerz ,

You are correct there is no rule to align `:=` in signal, constant and variable declarations.  

> Note that the rule constant_010 would actually have a potential to conflict with any rule that attempts to align the := in case the configuration was enforced to anything else than [...]

There could be a conflict.  I would need to check if a single space is added as part of the alignment rule, or if it will just align based on where the `:=` is currently to see if I could remove it.

> to enforce 1 space only.

There does seem to be a discrepancy between the documentation for `constant_010` and it's actual configuration.  I can update the documentation to match.

> This could fall into the aspect of reworking rules in a more global context as the principle would not only apply to constants, but also to signals (where default values should be avoided, but sometimes are OK) and variables.

I agree this issue could wait until #124 is completed.

> PS: The configuration of number_of_spaces seems also not working for:

Hmmm....after some testing you are correct.  I will investigate.

--Jeremy

jeremiah-c-leary, 2022-11-30:
Evening @staerz ,

I did a survey of where signal, constant and variable declarations are defined in the LRM:
[image]
Each `_item` production is called by a corresponding `_part`.  For example:
```text
   block_declarative_part ::=
       { block_declarative_item}
```
So I would use the group name `declarative_part` to handle all rules which cover multiple `declarative_item`s.  I should be able to do this fairly easily with my current data structure.  This would also help me address issue #887.

--Jeremy

staerz, 2022-11-30:
Hi @jeremiah-c-leary,

yes, if I understand it correctly, that looks like a right step in to the direction of abstracting the rules and grouping them more along the lines of the LRM.

You got :+1: :+1: from me :wink:

jeremiah-c-leary, 2022-12-02:
Morning @staerz ,

I pushed an update for this to the `issue-864` branch.  I think this was a good exercise to go through to help validate the new rule methodology.

There is a single rule `declarative_part_400` which will align `:=` in each declarative part.  I can see moving existing alignment rules such as `process_033` and blank line rules like `process_027` into the `declarative_part` group.

Let me know what you think of the update and whether we can close this issue.

Thanks,

--Jeremy

staerz, 2022-12-13:
Hi @jeremiah-c-leary,

I have successfully tested your updated and it works like intended.

Going over [the rules for keyword alignment](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_keyword_alignment_rules.html#configuring-keyword-alignment-rules), yes, I'd see potential for (re)moving some rules into the scope of the new `declarative_part_xxx` realm.
E.g. I'd group all rules together that check for the colon alignment (no matter if signal, constant, variable) and similarly, into `declarative_part_400`, those for the default value assignment.
You might open a 3rd of these rules for (inline) comments.

You might though considering simply noting this for a future major release as it might come to a surprise to some users to see many rules disappear/change.

jeremiah-c-leary, 2022-12-14:
Hey @staerz ,

> You might though considering simply noting this for a future major release as it might come to a surprise to some users to see many rules disappear/change.

Agree on waiting to consolidate existing rules.

I will get this merged to master.

--Jeremy

PR #895 (2022-12-14): Issue 864
Resolves #864 
Resolves #887
closed 2022-12-14 by jeremiah-c-leary