jeremiah-c-leary/vhdl-style-guide #866: Document decision process for rule generation

jeremiah-c-leary, 2022-11-12:
Guidelines for creating rules should be documented.  This will force an evaluation of current rules and whether they are properly named and scoped.  This could lead to a reduction in rule count and drive improvements into the code base.

jeremiah-c-leary, 2022-11-13:
Hey @staerz,

I pushed some [documentation](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-866/docs/theory_of_operation/rule_selection.rst) to the `issue-866` branch.  It is more of a stream of consciousness than a coherent document.

It could be a good starting point though.

Let me know what you think.

Regards,

--Jeremy

staerz, 2022-11-15:
Hey @jeremiah-c-leary,

I looked over it.

I would suggest to simplify your life a little:
- You could combine the whitespace rules into one when ever there is a longer list of elements and we'd usually only put exactly one space in between elements, i.e. in `*architecture* identifier *of* entity_name *is*` - I do not see why at any position where a space is allowed, anything else than exactly 1 space would/should be used. So I'd make that one rule, not 4.
- Similar for `*end* [ *architecture* ] [ architecture_simple_name ] ;` you could have 1 rule instead of 3
- For rules related to casing you might want to introduce a base rule: I'd **always** want `begin` and `end` to be the very same case, no matter what they begin or end. Similar with `is` and `of`.

PR #873 (2022-11-15): Adding theory of operation section.
Resolves #866
PR #880 (2024-02-02): Adding tokeninzing, tokens and classifying documentation.
Resolves #866
closed 2024-02-02 by jeremiah-c-leary