jeremiah-c-leary/vhdl-style-guide #867: Typo in "Configuring Concurrent Procedure Call Statement Rules" page

JHertz5, 2022-11-12:
Sorry if this is too small of an issue to raise, but I've noticed a typo on the following docs page [Configuring Concurrent Procedure Call Statement Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_concurrent_procedure_call_statement_rules.html)

**Environment**
Typo seen in Firefox browser on Windows 10 (not relevant).

**Describe the bug**
On the docs page [Configuring Concurrent Procedure Call Statement Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_concurrent_procedure_call_statement_rules.html), the table lists a "last_open_paren" option, but in reality, this option is called "last_close_paren".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "docs/configuring_concurrent_procedure_call_statement_rules.rst"
2. Scroll down to line 62.
3. See error: `last_open_paren`.

**Expected behavior**
This should say `last_close_paren`.

**Screenshots**
[image]

jeremiah-c-leary, 2022-11-12:
Morning @JHertz5 ,

> On the docs page [Configuring Concurrent Procedure Call Statement Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_concurrent_procedure_call_statement_rules.html), the table lists a "last_open_paren" option, but in reality, this option is called "last_close_paren".

You are correct.  Must have been a copy/paste error.

> I would be more than happy to fix this myself, but I don't want to go creating branches on your repository uninvited. Would it be ok for me to create a branch and pull request for this?

I would welcome any pull request.  BTW, thanks for opening an issue first.

Could you add the following text "Resolves #867" in the pull request description.  GitHub will link the pull request to the issue and I use it to manage my issues.

If you find anything else you would like to add a pull request for, just create an issue and then the pull request.

> PS: I'm think this project is fantastic, nice work!

Thanks for the kind words.

Regards,

--Jeremy

JHertz5, 2022-11-12:
@jeremiah-c-leary Thanks very much! The PR has been created.
I will make sure to follow that process if I find anything else.

Thanks again!
Jukka

jeremiah-c-leary, 2022-11-12:
Afternoon @JHertz5 ,

The pull request looks good, I will merge it in.

Thanks again,

--Jeremy

closed 2022-11-12 by jeremiah-c-leary