jeremiah-c-leary/vhdl-style-guide #869: Broken link on "Configuring Blank Lines" page.

JHertz5, 2022-11-12:
**Environment**
 VSG 3.13.1

**Describe the bug**
The link to rule if_030 is broken and takes the user to a non-existent URL.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [Configuring Blank Lines](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_blank_lines.html).
2. Scroll down to the "Rules Enforcing Blank Lines" heading.
3. Click on the "if_030" link.
4. See error: The link takes you to the non-existent URL [https://vhdl-style-guide.readthedocs.io/en/latest/if_statement_rules.html#if_statement-030](https://vhdl-style-guide.readthedocs.io/en/latest/if_statement_rules.html#if_statement-030)

**Expected behavior**
The link should take you to [if_030](https://vhdl-style-guide.readthedocs.io/en/latest/if_rules.html#if-030).

jeremiah-c-leary, 2022-11-13:
Morning @JHertz5,

Thanks for pointing out the invalid link.  Those have surprisingly been an issue for me.

I will merge this into master.

Thanks again,

--Jeremy

closed 2022-11-13 by jeremiah-c-leary

JHertz5, 2022-11-13:
Hi @jeremiah-c-leary.

No problem at all! I saw that you raised #871 regarding this type of issue. I'll avoid raising further issues on specific instances of broken URLs and let that issue cover all of them.

Thanks,
Jukka

jeremiah-c-leary, 2022-11-13:
Hey @JHertz5 ,

yeah, I decided to write some tests for my documentation to check the links.  I have found missing links and deprecated links.  All kinds of interesting things.  It is a slog to get through though.

--Jeremy

JHertz5, 2022-11-13:
@jeremiah-c-leary If I can help with repairing/updating links, let me know! I'd be more than happy to take on some of the "grunt work".