jeremiah-c-leary/vhdl-style-guide #878: Whitespace_007 being triggered in pragma

jeremiah-c-leary, 2022-11-21:
The whitespace_007 rule is being triggered in the following code:
```vhdl 
  5     --vhdl_comp_off
  6     sdaIobufE : component IOBUF
  7         generic map (
  8             DRIVE      => 12,
  9             IOSTANDARD => "DEFAULT",
 10             SLEW       => "SLOW"
 11         )
 12         port map (
 13             O  => i2cSdaIn,
 14             IO => i2cSdaxDIO,
 15             I  => i2cSdaOut,
 16             T  => i2cSdaOut
 17         );
 18     --vhdl_comp_on
```

closed 2022-11-21 by jeremiah-c-leary

Scafir, 2022-11-22:
Hey @jeremiah-c-leary ,
Thank you for going ahead with the issue and fix. I'm a bit overwhelmed at work and am sorry I did not get back on it earlier.
As always, thank you very much for your dedication

jeremiah-c-leary, 2022-11-22:
Hey @Scafir ,

> Thank you for going ahead with the issue and fix. I'm a bit overwhelmed at work and am sorry I did not get back on it earlier.
As always, thank you very much for your dedication

I completely understand.  Good luck at work and hopefully it calms down soon.

--Jeremy

PR #879 (2022-11-21): Issue #878:  Fixing whitespace_007 rule triggers in pragmas.
Resolves #878