jeremiah-c-leary/vhdl-style-guide #882: Move concurrent_007 rule to when_002

Scafir, 2022-11-23:
**Is your feature request related to a problem? Please describe.**
I believe it would make more sense to group concurrent_007:
> This rule checks for code after the else keyword.

with the existing `when_001` rule, as they are both relating to `when` statements.

**Describe the solution you'd like**
Move `concurrent_007` to `when_002`.

jeremiah-c-leary, 2023-11-19:
Morning @Scafir ,

Well, it looks like it has almost been a year since you raised this issue.  I decided to move the rule to `conditional_waveforms_001` as it seemed a better fit than `when_002`.  Eventually, `when_001` will be moved also.

Let me know what you think.

I will plan to merge this to master Dec 3rd if I do not hear from you.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-09:
Morning @Scafir ,

I will merge this into master.

Thanks,

--Jeremy

PR #1049 (2023-12-09): Issue#882:  Moving rule concurrent_007 to conditional_waveforms_001.
Resolves #882
closed 2023-12-09 by jeremiah-c-leary