jeremiah-c-leary/vhdl-style-guide #883: library_009 and comment_010 overlap

Scafir, 2022-11-24:
**Environment**
 CentOS Linux 7 (Core)
VHDL Style Guide (VSG) version: 3.13.1+zip.file

**Describe the bug**
`library_009` and `comment_010` are checking the same lines of code. If one sets a different indentation level for `library_009` it is then impossible to pass successfully style checking, as one of the two rules is guaranteed to fail

**To Reproduce**
Steps to reproduce the behavior:
1. Use `rules.yaml`:
```
rule:
  library_008:
    indentSize: 0
  library_009:
    indentSize: 0
```
`sample.vhd`:
```
library ieee;
  --! For logic elements
use ieee.std_logic_1164.all;
--! For using natural type
use ieee.numeric_std.all;
```
3. Run `vsg -c rules.yaml -f sample.vhd`
4. See that both `library_009` and `comment_010` are triggered

**Expected behavior**
Two possible solutions:
* `comment_010` should not check lines that are already checked by `library_009`, or
* `library_009` should be removed and `comment_010` respect indent level of `library_008

jeremiah-c-leary, 2022-12-15:
Evening @Scafir ,

It took a bit, but I believe I have a solution for this.  I added `use_clause` to the indent configuration and added a special option `token_after_library_clause` which can be used to configure the indenting of `use_clause`s.

If you use the following configuration, you will get the indenting you want:
```yaml
   indent:
    tokens:
      use_clause:
        keyword:
          token_after_library_clause: current
```
There is [indent documentation](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/issue-883/docs/configuring_indentation.rst) which describes how VSG sets indents.

Let me know if this fixes the issue for you.

Regards,

--Jeremy

Scafir, 2022-12-21:
Hi @jeremiah-c-leary,

I am slightly confused. I am using the version available in branch `issue-883` and:
* Running vsg with your configuration at top level, an error is returned with `KeyError: 'use_clause'`.
* Running `vsg -oc tmp.json`, I do not see a `use_clause` entry in the indent section.
Am I doing something wrong? I am not 100% sure that I am entering your configuration in the correct place.

In any case, this fix should fine enough for me, but would probably confuse new users.. Is there a way for the indentation level set in library_009 to be automatically applied to the indentation configuration?

Regards,

Clyde

jeremiah-c-leary, 2022-12-21:
Morning @Scafir ,

Is it possible you are calling a different vsg than the one in branch `issue-883`?  I just ran into this when trying to validate `use_clause` was not listed in the configuration because I have an alias for vsg which was pointing to another working copy.  When I issued `vsg -oc tmp.json` in the correct working copy I see `use_clause`.
[image]

I double checked my working copy is up to date and I see the change on github:
[image]

> In any case, this fix should fine enough for me, but would probably confuse new users.. Is there a way for the indentation level set in library_009 to be automatically applied to the indentation configuration?

I will assume you are referring to the `indentSize` option for `library_009`:
```json
     "library_009": {
       "disable": false,
       "fixable": true,
       "indentSize": 2,
       "indentStyle": "spaces",
       "phase": 4,
       "severity": "Error"
     },
```
The indenting is split between a level and the number of spaces per level (`indentSize`).  Setting `indentSize` to 0 will push the `use_clause` all the way to the left because vsg multiplies the indent level and `indentSize` to determine the amount of leading whitespace.  I think if I allowed `library_009` to adjust the indent level, it would violate the separation of the indent level and `indentSize`.

> but would probably confuse new users...

I really should have set the indent of a `use_clause` to be `current` by default, so it would have the same indent as the `library_clause`.   I would like to change it, but I worry about the affect on users.

Regards,

--Jeremy

Scafir, 2023-01-05:
Hi @jeremiah-c-leary,

I found the issue with the `KeyError: 'use_clause'`. I was calling the correct vsg binary, but it was still executing code contained in a previous installation, which is something that did not happen when I checked other issues. Probably something to do with path shadowing.

In any case, I was able to test your fix and I am happy with it.

> I really should have set the indent of a use_clause to be current by default, so it would have the same indent as the library_clause. I would like to change it, but I worry about the affect on users.

It is a very reasonable explanation. If the 4.0 rewrite happens, it'll probably be a good occasion to change that.

Thank you!
Clyde

Scafir, 2023-01-05:
I closed the issue, but the issue-883 branch still needs merging! Sorry about that

PR #900 (2023-04-23): Issue #883: Fixing conflict issue with library_009 and comment_010.
Resolves #883
closed 2023-04-23 by jeremiah-c-leary