jeremiah-c-leary/vhdl-style-guide #884: Named association after parenthesis in procedure call with procedure_call_401

svnesbo, 2022-11-25:
**Environment**
Ubuntu 22.04.
VSG: 3.1.13 (pip)

**Describe the bug**
When using named association in a procedure call, where the first parameter is on the same line as the call and opening parenthesis, the procedure_call_401 fails to indent properly and inserts whitespace before the first argument.

**To Reproduce**
Example file:

```
library ieee;
  use ieee.std_logic_1164.all;

entity test is
  port (
    clk : in    std_logic;
    x   : in    std_logic;
    y   : in    std_logic;
    z   : out   std_logic
  );
end entity test;

architecture rtl of test is

  procedure my_procedure (
    constant a            : in  std_logic;
    constant b            : in  std_logic;
    signal output_param_c : out std_logic)
  is
  begin

    c <= a and b;

  end procedure my_procedure;

begin

  p_test : process (clk) is
  begin

    if rising_edge(clk) then
      my_procedure(a => x,
                   b => y,
                   c => z);
    end if;

  end process p_test;

end architecture rtl;
```

Steps to reproduce the behavior:
`vsg -f test.vhd --fix`

**Expected behavior**
Not expecting any changes to the file in this case.

**Observed behavior**
Whitespace inserted between opening parenthesis and the first argument to the procedure:

```
my_procedure(                  a => x,
                   b => y,
                   c => z);
```

Output from vsg:
```
================================================================================
File:  test.vhd
================================================================================
Phase 2 of 7... Reporting
Total Rules Checked: 256
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  whitespace_005            | Error      |         32 | Remove spaces after open (.
----------------------------+------------+------------+--------------------------------------
```

**Additional context**
Occurs when vsg is called with no custom configuration (default config used).

It is the whitespace_005 rule that complains in the output of vsg, but I think it is the procedure_call_401 rule that is at fault. If I run vsg with procedure_call_401 disabled then it does not happen:

procedure.yaml:
```
rule:
  procedure_call_401:
    disable: False
```

`vsg -c procedure.yaml -f test.vhd --fix`
No changes in indentation for the procedure then (but then it also won't fix indenting in procedure calls).

jeremiah-c-leary, 2022-11-25:
Good Afternoon @svnesbo ,

I found the issue and pushed an update to the `issue-884` branch.  When you get a chance could you validate it on your end and let me know if it addresses the issue for you.  If it does then I will merge the branch to master and close this issue.

Thanks,

--Jeremy

svnesbo, 2022-11-28:
Hi Jeremy,

Thanks for looking into this :smile: 

I tried the `issue-884` branch and it seems to have fixed the issue I had :+1: 

But since the code in my example was already nicely formatted I tried adding some combinations of whitespace/wrong-indenting to that procedure call to see that it would be able to fix it. It works most of the time, but I found one case that it doesn't handle:

```
      my_procedure(a =>   x,
                   b =>     y,
                   c =>       z);
```
If I add some whitespace after `=>` in my example (shown above), then vsg is not able to fix it. No reported warnings/errors, and no changes to the source file with `--fix`. But note that I ran vsg with default configuration in this case, and I haven't checked if there are other config options that needs to be enabled to fix this.

jeremiah-c-leary, 2022-11-28:
Morning @svnesbo ,

> I tried the issue-884 branch and it seems to have fixed the issue I had

Awesome.

> But since the code in my example was already nicely formatted I tried adding some combinations of whitespace/wrong-indenting to that procedure call to see that it would be able to fix it. It works most of the time, but I found one case that it doesn't handle:

There is currently not a rule similar to `port_map_007` for procedure calls.  Would you be willing to open another issue since the original bug was fixed?  This will help me in tracking bugs versus adding new rules.

Thanks,

--Jeremy

svnesbo, 2022-11-29:
Thanks for the quick fix :+1: 

> There is currently not a rule similar to port_map_007 for procedure calls. Would you be willing to open another issue since the original bug was fixed? This will help me in tracking bugs versus adding new rules.

I opened a new issue for this here: #886

jeremiah-c-leary, 2022-11-29:
Awesome, I will get this issue merged to master.

--Jeremy

PR #885 (2022-11-29): Rule procedure_call_401 was adding addition whitespace when not needed.
Resolves #884
closed 2022-11-29 by jeremiah-c-leary