jeremiah-c-leary/vhdl-style-guide #886: Rule to check for single space after => operator in procedure calls

svnesbo, 2022-11-29:
**Environment**
Ubuntu 22.04.
VSG: 3.1.13 (branch issue-884)

**Is your feature request related to a problem? Please describe.**
The problem is with procedure calls where there are unnecessary spaces after the `=>`, for example:
```
    my_procedure(a => x,
                 bx   =>  y,
                 c      =>      z);
```

When running VSG with the default rules (e.g. `vsg -f test.vhd --fix`), the procedure parameters and `=>` are correctly aligned and the unnecessary spaces **before** `=>` are removed. But the unnecessary spaces **after** the `=>` operator is not removed. Output after running VSG:
```
    my_procedure(a  => x,
                 bx =>  y,
                 c  =>      z);
```

**Describe the solution you'd like**
Removal of unnecessary spaces after the `=>` operator and alignment of the parameters after the `=>` operator.

According to @jeremiah-c-leary (in #884) a rule similar to `port_map_007` is needed for procedure calls.

**Additional context**
Full example file that can be used for testing:
```
library ieee;
use ieee.std_logic_1164.all;

entity test is
  port (
    x : in    std_logic;
    y : in    std_logic;
    z : out   std_logic
  );
end entity test;

architecture rtl of test is

  procedure my_procedure (
    constant a  : in  std_logic;
    constant bx : in  std_logic;
    signal c    : out std_logic)
  is
  begin
    c <= a and b;
  end procedure my_procedure;

begin

  p_test : process (all) is
  begin
    my_procedure(a => x,
                 bx   =>  y,
                 c      =>      z);
  end process p_test;

end architecture rtl;

```

jeremiah-c-leary, 2022-11-29:
Morning @svnesbo ,

I pushed an update for this to the `issue-886` branch.  I added rule `procedure_call_101` which will enforce a single space after the `=>`.  When you get a chance could you validate it on your end and let me know if there are any issues?

Thanks,

--Jeremy

jeremiah-c-leary, 2022-11-29:
Hey @svnesbo ,

Awesome, I will get this merged to master.

--Jeremy

PR #889 (2022-11-29): Issue #886 : Adding rule procedure_call_101
Resolves #886
closed 2022-11-29 by jeremiah-c-leary