jeremiah-c-leary/vhdl-style-guide #887: whitespace_011 can break alignment of default assignment

svnesbo, 2022-11-29:
**Environment**
Ubuntu 22.04.
VSG: 3.1.13 (branch issue-884)

**Describe the bug**
When there are several signal or constant declarations with default assignments, where the assignments are already aligned, but one of the constants/signals has for instance a range specifier which includes a math operator with no spaces around the operator, it appears that the whitespace_011 rule will break the alignment when it adds whitespace around the math operator.

**To Reproduce**
Example input file:
```
library ieee;
use ieee.std_logic_1164.all;

entity test is
end entity test;

architecture rtl of test is

  constant x      : integer                  := 1024;
  constant abc    : integer                  := 2048;
  constant abcdef : integer range 0 to abc/2 := 0;

begin

end architecture rtl;
```

Steps to reproduce the behavior:
1. Run VSG with default config on the input file: `vsg -f test.vhd --fix`
2. After running VSG, the assignment to the constant `abcdef` in the example is no longer aligned.

Output after running VSG:
```
library ieee;
  use ieee.std_logic_1164.all;

entity test is
end entity test;

architecture rtl of test is

  constant x      : integer                  := 1024;
  constant abc    : integer                  := 2048;
  constant abcdef : integer range 0 to abc / 2 := 0;

begin

end architecture rtl;
```

**Expected behavior**
Alignment of assignments is run after whitespace has been added around the math operator.

jeremiah-c-leary, 2022-11-29:
Morning @svnesbo ,

This is related to issue #864.  I do not have a rule yet that aligns the `:=`.  When the spaces are added around the `/` it pushes the `:=` to the right. 

I will  move this and issue #864 up the priority list.

Regards,

--Jeremy

svnesbo, 2022-11-29:
Hi Jeremy,

Sorry for spamming you with all these issues, as you can probably tell I'm setting up VSG for some existing code and ran into some of these issues in the process :smile: 

I think I looked at issue #864 earlier. It looked related, but not precisely the same, so I thought it made sense to submit another issue.

But I think it's probably not the worst issue to be honest, since I guess it can never offset alignment by more than 2 characters.

jeremiah-c-leary, 2022-11-29:
Morning @svnesbo ,

> Sorry for spamming you with all these issues, as you can probably tell I'm setting up VSG for some existing code and ran into some of these issues in the process 😄

Not a problem.  This is typical.  A flurry of new issues as someone tries VSG on their code base which exposes some issues, then  it quiets down again.  I appreciate you letting me know of the issues you are encountering.

> But I think it's probably not the worst issue to be honest, since I guess it can never offset alignment by more than 2 characters.

It would depend on how many other whitespace rules would be applied before the assignment.

--Jeremy

jeremiah-c-leary, 2022-12-02:
Morning @svnesbo ,

I pushed a fix for this to the `issue-864` branch.  When you get a chance could you validate on your end that it is working.  If it is then I will close this ticket and merge it to master after issue #864 is also resolved.

Thanks,

--Jeremy

svnesbo, 2022-12-05:
Hi Jeremy,

Thanks for the fix and sorry for the late reply.

I tested the `issue-864` branch now. It fixes the issue with whitespace around math operators that I had reported here.

---

Maybe this should go in a separate issue, but I noticed that it now wants to align things more aggressively. Here is an example file:

```
library ieee;
use ieee.std_logic_1164.all;

entity test is
end entity test;

architecture rtl of test is

  constant CX : string := "Test";
  
  type my_type1 is (A, B, C, D);
  type my_type1 is (ABC, DEF, GHI, JKL);
  
  constant x      : integer                  := 1024;
  constant abc    : integer                  := 2048;
  constant abcdef : integer range 0 to abc / 2 := 0;

begin

end architecture rtl;
```

Previously (using vsg 3.13.1 installed with pip) the output when I run `vsg -f test.vhd --fix` looked like this:
```
library ieee;
  use ieee.std_logic_1164.all;

entity test is
end entity test;

architecture rtl of test is

  constant cx : string := "Test";

  type my_type1 is (a, b, c, d);

  type my_type1 is (abc, def, ghi, jkl);

  constant x      : integer                  := 1024;
  constant abc    : integer                  := 2048;
  constant abcdef : integer range 0 to abc / 2 := 0;

begin

end architecture rtl;
```

With the `issue-864` branch it wants to align `constant CX : string := "Test";` with the lines much further below.

```
library ieee;
  use ieee.std_logic_1164.all;

entity test is
end entity test;

architecture rtl of test is

  constant cx : string                         := "Test";

  type my_type1 is (a, b, c, d);

  type my_type1 is (abc, def, ghi, jkl);

  constant x      : integer                    := 1024;
  constant abc    : integer                    := 2048;
  constant abcdef : integer range 0 to abc / 2 := 0;

begin

end architecture rtl;
```

It seems that it's the new rule for issue 864 that is responsible:
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  declarative_part_400      | Error      |          9 | Move := 24 columns
----------------------------+------------+------------+--------------------------------------
```

From what I can see in the docs it is supposed to respect the keyword alignment rules. But `blank_line_ends_group` is supposed to be true by default, and I ran without any custom rules here, so it looks like it didn't respect `blank_line_ends_group`.

jeremiah-c-leary, 2022-12-06:
Morning @svnesbo ,

I checked the configuration and `blank_line_ends_group` was being set to `False` in the rule definition.  I changed it back to `True` and pushed an update to the `issue-864` branch.  Let me know if it is working for you.

Thanks,

--Jeremy

PR #895 (2022-12-14): Issue 864
Resolves #864 
Resolves #887
closed 2022-12-14 by jeremiah-c-leary