jeremiah-c-leary/vhdl-style-guide #888: Keyword alignment rules do not work for process_035

svnesbo, 2022-11-29:
**Environment**
Ubuntu 22.04.
VSG: 3.1.13 (branch issue-884)

**Describe the bug**
The rule process_035 aligns inline comments between begin and end keywords in a process.
[According to the documentation](https://vhdl-style-guide.readthedocs.io/en/latest/process_rules.html#process-035) the [Keyword Alignment Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_keyword_alignment_rules.html) can be used to fine-tune process_035.

But in my experience, process_035 aligns **all** inline comments in a process regardless of any keyword alignment rules.

**To Reproduce**
Example input file below.

To better illustrate the issue, one of the lines in the process is rather long and has an inline comment. It would be undesirable to have all inline comments aligned to it.

```
library ieee;
  use ieee.std_logic_1164.all;

entity test is
  port (
    x : in    std_logic;
    y : in    std_logic;
    z : out   std_logic
  );
end entity test;

architecture rtl of test is
  constant c_constant_with_a_really_really_really_long_name : std_logic := '1';
  signal a : std_logic;
  signal b : std_logic;
begin

  p_test : process (all) is
  begin
    if (x = c_constant_with_a_really_really_really_long_name) then         -- Comment 1
      z <= '0';
    else -- Comment 2
      z <= '1';     -- Another comment
      a <= '0';   -- And another comment
      b <= '1';        -- Last comment
    end if;
  end process p_test;

end architecture rtl;
```

Steps to reproduce the behavior:
1. Run VSG with the default rules (e.g. vsg -f test.vhd --fix)
2. The result after running VSG has all inline comments in the process aligned to the long line (see below)

Result:
```
library ieee;
  use ieee.std_logic_1164.all;

entity test is
  port (
    x : in    std_logic;
    y : in    std_logic;
    z : out   std_logic
  );
end entity test;

architecture rtl of test is

  constant c_constant_with_a_really_really_really_long_name : std_logic := '1';
  signal   a                                                : std_logic;
  signal   b                                                : std_logic;

begin

  p_test : process (all) is
  begin

    if (x = c_constant_with_a_really_really_really_long_name) then         -- Comment 1
      z <= '0';
    else                                                                   -- Comment 2
      z <= '1';                                                            -- Another comment
      a <= '0';                                                            -- And another comment
      b <= '1';                                                            -- Last comment
    end if;

  end process p_test;

end architecture rtl;

```

**Expected behavior**
The expected result would perhaps be something like this:
```
    if (x = c_constant_with_a_really_really_really_long_name) then  -- Comment 1
      z <= '0';
    else         -- Comment 2
      z <= '1';  -- Another comment
      a <= '0';  -- And another comment
      b <= '1';  -- Last comment
    end if;
```

There is a [Keyword Alignment Rule](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_keyword_alignment_rules.html#rule-specific-keyword-alignment-configuration) called `if_control_statements_ends_group` that is enabled by default, which it appears from the examples is supposed to limit alignment to within if/else blocks.

But this rule does not appear to be active.

I have tried explicitly enabling it with a config like this (and running VSG with -c cfg.yaml):
```
rule:
  process_035: # Alignment of inline comments between the process begin and end process lines.
    disable : false
    if_control_statements_ends_group : true
```

But this makes no difference. I have done some quick testing with other keyword alignment rules, adding some blank lines etc., but it seems like no matter what I do the alignment is done across the entire process.

**Additional context**
I am not 100% sure that this is a bug, it is possible that it's a configuration issue on my side, but so far I have not been able to get it to work as expected.

jeremiah-c-leary, 2022-12-07:
Good Evening @svnesbo ,

There is a single sentence the in [Configuring Keyword Alignment Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_keyword_alignment_rules.html) which states:

[image]

So all of the configuration options are not available in all keyword alignment rules.  It is not at all obvious though.  Any suggestions to make it stand out would be appreciated.  The entire page could use a refresh.

If you use the `-rc` option on `process_035`:
```mono
$ vsg -rc process_035
{
  "rule": {
    "process_035": {
      "indentStyle": "spaces",
      "indentSize": 2,
      "phase": 5,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "compact_alignment": false,
      "blank_line_ends_group": false,
      "comment_line_ends_group": false,
      "separate_generic_port_alignment": true,
      "include_lines_without_comments": true
    }
  }
}
```
It will give you all of the configuration parameters for the rule.  In this case `if_control_statements_ends_group` is not available for this rule.

However, one could argue `separate_generic_port_alignment` should not be included.

I guess the question now is, would like `if_control_statements_ends_group` included or just update the documentation to make it clear the option is not always available?

Regards,

--Jeremy

jeremiah-c-leary, 2023-04-01:
Hey @svnesbo,

Just wanted to ping you on this issue to see if it is still a problem.  I will plan to close this issue April 15th if I do not hear back from you.

Regards,

--Jeremy

closed 2023-04-19 by jeremiah-c-leary

svnesbo, 2023-05-25:
Hi Jeremiah,

Thanks for the detailed reply, and sorry for the very late response.

Maybe "bug" was the wrong label for this issue since it technically works as intended then. I think it probably makes sense to include `if_control_statements_ends_group` in `process_035` though, since `if` statements mostly appear in processes.

> There is a single sentence the in [Configuring Keyword Alignment Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_keyword_alignment_rules.html) which states:
> "Some of the configurations are available in all keyword alignment rules, while others are rule specific."

In your reply you showed me how to use `-rc` to check which (keyword alignement) rules apply to `process_035`. Perhaps it would be a good idea to mention this after that sentence? Unless you feel that would be to duplicate information that's already found elsewhere. I think the documentation for VSG is very extensive (which is awesome!), but a consequence is that most people (such as myself) probably won't read it all of it. I don't think I knew about the `-rc` option (or at least I didn't think of checking the rule that way).

As I'm writing this it strikes me that perhaps the terminology in the documentation is a bit inconsistent, so there is one last thing I want to mention. It can sometimes be a bit confusing to read the documentation since so many things are called a "rule".  A rule like `if_control_statements_ends_group` is sometimes referred to as a "keyword alignment rule" (or just a rule for short), a configuration, or attribute. Unless I'm mistaken, attribute seems like it's the right terminology according to the documentation:
https://vhdl-style-guide.readthedocs.io/en/latest/configuring_overview.html#rule
I guess that's not exactlly a 5 minute fix, but I think it's worth mentioning.

Best regards,

Simon