jeremiah-c-leary/vhdl-style-guide #890: Parsing error for procedure definition w/ 2D array/vector and default assignment

svnesbo, 2022-11-29:
**Environment**
Ubuntu 22.04.
Python 3.8.13
VSG: 3.1.13 (latest master https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/8994593af99a541111531f35f10851cd7da37d3b)

**Describe the bug**
When a parameter to a procedure has a type of array of std_logic_vector and the parameter has a default assignment such as `(others => (others => '0'))`, then the parser complains that the last closing `)` is unexpected.

Note that such an assignment fails when it is for a procedure parameter, but is accepted as a normal signal assignment (and probably fine for variables and constants as well).

**To Reproduce**

Example input file:
```
library ieee;
use ieee.std_logic_1164.all;

entity test is
end entity test;

architecture rtl of test is
  type t_slv_array is array (natural range <>) of std_logic_vector;

  signal x : t_slv_array(0 to 1)(7 downto 0) := (others => (others => '0'));
  signal y : t_slv_array(0 to 1)(7 downto 0);
  
  procedure my_procedure (
    constant a : in  t_slv_array(0 to 1)(7 downto 0) := (others => (others => '0'));
    signal b   : out t_slv_array(0 to 1)(7 downto 0)
    )
  is
  begin
    b <= a;
  end procedure my_procedure;

begin

  p_test : process (all) is
  begin
    my_procedure(a => x,
                 b => y);
  end process p_test;

end architecture rtl;

```

Steps to reproduce the behavior:
1. Run `vsg -f test.vhd --fix` on the test file
2. VSG will complain that it found `)` when it's expecting `begin`:

```
Error: Unexpected token detected while parsing architecture_body @ Line 14, Column 83 in file None
       Expecting : begin
       Found     : )
```

**Expected behavior**
It should be able to parse the code.

Also, the error printed by the parser outputs **None** for the filename.

**Additional context**
Note that I have not actually tried to compile the code example. But I believe it should be valid VHDL, and it is based on a real piece of code with a procedure with such a parameter (which failed to parse for the same reason).

jeremiah-c-leary, 2022-11-29:
Morning @svnesbo ,

I pushed an update for this issue to the `issue-890` branch.  There was also an issue with signals, but variables would have been fine.  When you get a chance let me know if this resolves the issue on your end.

Thanks,

--Jeremy

jeremiah-c-leary, 2022-11-29:
Sweet....merging to master.

--Jeremy

PR #891 (2022-11-29): Issue #890 : Fixing parsing error for constants and signals with defa…
Resolves #890
closed 2022-11-29 by jeremiah-c-leary