jeremiah-c-leary/vhdl-style-guide #892: Empty list of oLines.lParens in check_middle_line()

makoeppel, 2022-11-29:
**Environment**
Arch Linux 5.15.78-1-lts 
VHDL Style Guide (VSG) version: 3.13.1+zip.file
Python 3.10.8

**Describe the bug**
In the function `check_middle_line()` the line `oLines.iNextIndent = oLines.lParens[-1].iExpectedColumn` fails because `oLines.lParens` is an empty list.

**To Reproduce**
I have the following VHDL file in my project.

```vhdl
---------------------------------------------------------------------------------
--
--   Copyright 2017 - Rutherford Appleton Laboratory and University of Bristol
--
--   Licensed under the Apache License, Version 2.0 (the "License");
--   you may not use this file except in compliance with the License.
--   You may obtain a copy of the License at
--
--       http://www.apache.org/licenses/LICENSE-2.0
--
--   Unless required by applicable law or agreed to in writing, software
--   distributed under the License is distributed on an "AS IS" BASIS,
--   WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
--   See the License for the specific language governing permissions and
--   limitations under the License.
--
--                                     - - -
--
--   Additional information about ipbus-firmare and the list of ipbus-firmware
--   contacts are available at
--
--       https://ipbus.web.cern.ch/ipbus
--
---------------------------------------------------------------------------------

--  Dave Sankey May 2013

library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity udp_rxram_shim is
  generic (
    bufwidth : natural := 0
  );
  port (
    mac_clk    : in    std_logic;
    rst_macclk : in    std_logic;
    --
    rxram_end_addr  : in    std_logic_vector(12 downto 0);
    rxram_send      : in    std_logic;
    rxram_write_buf : in    std_logic_vector(bufwidth - 1 downto 0);
    --
    rxram_req_send : in    std_logic;
    rxram_send_buf : in    std_logic_vector(bufwidth - 1 downto 0);
    --
    rxram_busy : in    std_logic;
    --
    rxram_end_addr_x : out   std_logic_vector(12 downto 0);
    rxram_send_x     : out   std_logic;
    rxram_sent       : out   std_logic
  );
end entity udp_rxram_shim;

architecture simple of udp_rxram_shim is

  type address_buf is array (2 ** bufwidth - 1 downto 0) of std_logic_vector(12 downto 0);

  signal end_address_buf : address_buf;
  signal last_busy       : std_logic;

begin

  input_block : process (mac_clk) is
  begin

    if rising_edge(mac_clk) then
      if (rst_macclk = '1') then
        end_address_buf <= (Others => (Others => '0'));
      elsif (rxram_send = '1') then
        end_address_buf(to_integer(unsigned(rxram_write_buf))) <= rxram_end_addr;
      end if;
    end if;

  end process input_block;

  output_block : process (mac_clk) is
  begin

    if rising_edge(mac_clk) then
      if (rxram_req_send = '1') then
        rxram_send_x <= '1'
        -- pragma translate_off
                        after 4 ns
        -- pragma translate_on
                        ;
        rxram_end_addr_x <= end_address_buf(to_integer(unsigned(rxram_send_buf)))
        -- pragma translate_off
                            after 4 ns
        -- pragma translate_on
                            ;
      else
        rxram_send_x <= '0'
        -- pragma translate_off
                        after 4 ns
        -- pragma translate_on
                        ;
        rxram_end_addr_x <=
        (
          Others => '0'
        )
          -- pragma translate_off
          after 4 ns
          -- pragma translate_on
          ;
      end if;
    end if;

  end process output_block;

  busy_block : process (mac_clk) is
  begin

    if rising_edge(mac_clk) then
      if ((rxram_busy = '0') and (last_busy = '1')) then
        rxram_sent <= '1'
        -- pragma translate_off
                      after 4 ns
        -- pragma translate_on
                      ;
      else
        rxram_sent <= '0'
        -- pragma translate_off
                      after 4 ns
        -- pragma translate_on
                      ;
      end if;
      last_busy <= rxram_busy
      -- pragma translate_off
                   after 4 ns
      -- pragma translate_on
                   ;
    end if;

  end process busy_block;

end architecture simple;
```
And I use this style file:

```yaml
rule:
  global:
    indentSize: 2
  entity_008:
    case: lower
  signal_007:
    disable: True
  port_012:
    disable: True
  instantiation_034:
    disable: True
      #signal_008:
      #disable: False
  port_603:
    disable: False
  port_604:
    disable: False
  port_011:
    disable: False
```
When I run `vsg --configuration style.yaml -f udp_rxram_shim.vhd --fix` I am getting the error:

```
Traceback (most recent call last):
  File "/home/makoeppel/.local/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "/home/makoeppel/.local/lib/python3.10/site-packages/vsg/__main__.py", line 131, in main
    fStatus, testCase, dJsonEntry, sOutputStd, sOutputErr = f((iIndex, sFileName))
  File "/home/makoeppel/.local/lib/python3.10/site-packages/vsg/apply_rules.py", line 103, in apply_rules
    oRules.fix(
  File "/home/makoeppel/.local/lib/python3.10/site-packages/vsg/rule_list.py", line 158, in fix
    oRule.fix(self.oVhdlFile, dFixOnly)
  File "/home/makoeppel/.local/lib/python3.10/site-packages/vsg/rule.py", line 104, in fix
    self.analyze(oFile)
  File "/home/makoeppel/.local/lib/python3.10/site-packages/vsg/rule.py", line 141, in analyze
    self._analyze(lToi)
  File "/home/makoeppel/.local/lib/python3.10/site-packages/vsg/rules/multiline_array_alignment.py", line 68, in _analyze
    self.calculate_expected_indents(oToi, oLines)
  File "/home/makoeppel/.local/lib/python3.10/site-packages/vsg/rules/multiline_array_alignment.py", line 86, in calculate_expected_indents
    dExpectedIndent = self.analyze_align_left_false_align_paren_true(oToi, oLines)
  File "/home/makoeppel/.local/lib/python3.10/site-packages/vsg/rules/multiline_array_alignment.py", line 112, in analyze_align_left_false_align_paren_true
    check_middle_line(oLine, oLines, self.indentSize)
  File "/home/makoeppel/.local/lib/python3.10/site-packages/vsg/rules/multiline_array_alignment.py", line 162, in check_middle_line
    oLines.iNextIndent = oLines.lParens[-1].iExpectedColumn
IndexError: list index out of range
```

**Fix**
Adding `if not oLines.no_parens():` similar to the check_first_line() function.

jeremiah-c-leary, 2022-11-30:
Evening @makoeppel ,

It seems to be related to the location of the semicolon on the assignment to rxram_end_addr_x:

```vhdl
rxram_end_addr_x <=
        (
          Others => '0'
        )
          -- pragma translate_off
          after 4 ns
          -- pragma translate_on
          ;
```

If the semicolon is on the same line as the closing parenthesis then is does not error out.

I am investigating.

--Jeremy

makoeppel, 2022-11-30:
Hi @jeremiah-c-leary,

thank you for looking to this.

In general your fix works but I think its cleaner to just `if not oLines.no_parens():`. The `no_parens()` already checks if there are no more paren. So the fix would be just:

```python
def check_middle_line(oLine, oLines, iIndentStep):
    set_expected_indent_of_line(oLine, oLines, iIndentStep)
    oLines.update_parens(oLine, iIndentStep)
    if not oLines.no_parens():
        oLines.iNextIndent = oLines.lParens[-1].iExpectedColumn
```
No usage of the new function `is_line_after_last_paren(oLines)` is needed.

One general remark: the name / usage of your new function is a bit odd. It says `is_line_after_last_paren(oLines)` and it returns `True` if there is no line after last paren ;)

```python
def is_line_after_last_paren(oLines):
    if len(oLines.lParens) == 0:
        return True
    return False
```

jeremiah-c-leary, 2022-11-30:
Morning @makoeppel ,

Yes, using `oLines.no_parens()` is a better solution than `is_line_after_last_paren`.  I have made the change and pushed an updated version to the `issue-892` branch.

Thanks,

--Jeremy

closed 2022-12-01 by jeremiah-c-leary