jeremiah-c-leary/vhdl-style-guide #896: Ability to read VHDL file from standard input

SethGower, 2022-12-07:
The ability to be able to read the VHDL file to be checked from standard input would make it easier/faster to be able to use VSG in editors asynchronously/without needing to save to be able to get diagnostics. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Specifically for neovim, it's possible to wrap arbitrary programs with a plugin and provide diagnostics based on the output of the program. I believe this is the purpose the `syntastic` output format exists for this program. I have done this with [null-ls](https://github.com/jose-elias-alvarez/null-ls.nvim)

jeremiah-c-leary, 2022-12-14:
Evening @SethGower ,

What is the best way for me to validate the pull request?

--Jeremy

SethGower, 2022-12-14:
@jeremiah-c-leary,
So I have been testing it in my day to day using [`null-ls`](https://github.com/jose-elias-alvarez/null-ls.nvim) and the [custom source](https://github.com/SethGower/dotfiles/blob/main/config/nvim/lua/plugins/null-ls.lua) I defined to wrap around VSG. But basic testing could just be done by simply invoking vsg and passing in the `--stdin` flag and send a file to the `stdin` of the program (like `vsg --stdin < file.vhd`). 

I just realized a small issue that I will need to patch. If you do both `--stdin` and `--fix` it'll fix the file, but then it'll write it to a file `stdin` in the current directory. So I can take a look at that later tonight probably. The expected behavior (I think) should be it doesn't attempt to fix the file read from standard input. Standard input should _only_ be used for checking (such as how I am using it in an editor).

SethGower, 2022-12-14:
For example with the following `test.vhd`
```vhdl
-- library ieee;
--     use ieee.std_logic_1164.all;
--     use ieee.numeric_std.all;

library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;
    use ieee.std_logic_unsigned.all;

entity test is
    port (
        i_clk : in    std_logic;
        i_a   : in    std_logic;
        o_x   : out   std_logic
    );
end entity test;

architecture behavioral of test is

begin

    o_x_pro : process (i_clk) is
    begin

        if rising_edge(i_clk) then
            o_x <= i_a;
        end if;

    end process o_x_pro;

end architecture behavioral;
```

If I run `vsg --stdin < test.vhd` I get the following output
```
================================================================================
File:  stdin
================================================================================
Phase 4 of 7... Reporting
Total Rules Checked: 434
Total Violations:    14
  Error   :    14
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  library_008               | Error      |          6 | Use 2 spaces for indent
  library_008               | Error      |          7 | Use 2 spaces for indent
  library_008               | Error      |          8 | Use 2 spaces for indent
  port_002                  | Error      |         11 | Use 2 spaces for indent
  port_004                  | Error      |         12 | Use 4 spaces for indent
  port_004                  | Error      |         13 | Use 4 spaces for indent
  port_004                  | Error      |         14 | Use 4 spaces for indent
  port_015                  | Error      |         15 | Use 2 spaces for indent
  process_001               | Error      |         22 | Use 2 spaces for indent
  process_003               | Error      |         23 | Use 2 spaces for indent
  if_001                    | Error      |         25 | Use 4 spaces for indent
  sequential_001            | Error      |         26 | Use 6 spaces for indent
  if_014                    | Error      |         27 | Use 4 spaces for indent
  process_006               | Error      |         29 | Use 2 spaces for indent
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

jeremiah-c-leary, 2022-12-15:
Morning @SethGower ,

> But basic testing could just be done by simply invoking vsg and passing in the --stdin flag and send a file to the stdin of the program (like vsg --stdin < file.vhd).

Thanks, I will give it a try.

> The expected behavior (I think) should be it doesn't attempt to fix the file read from standard input. Standard input should only be used for checking (such as how I am using it in an editor).

Wouldn't you also like the ability to fix the file?  I use VIM and I use the macro documented and I can just write my code without caring about formatting and then just hit the F9 key to fix it all.

--Jeremy

PR #897 (2023-09-08): Adds ability to read from standard input
Allows the user to send the VHDL file to VSG from `stdin` instead of reading files on the disk. Resolves #896
closed 2023-09-08 by jeremiah-c-leary