jeremiah-c-leary/vhdl-style-guide #901: Multiline Indent Rules: behaviour when first_paren_new_line

Scafir, 2022-12-21:
**Environment**
Fedora 37, VHDL Style Guide (VSG) version: 3.14.0+zip.file

**Describe the bug**
It does not seem to be possible to obtain the following:
```
wr_data <= (
  0,
  65535,
  32768
);
```

**To Reproduce**
Let's take `concurrent_401` as an example.
Following documentation we get the following rules:
```
# sampleRules.yaml
rule:
  concurrent_012:
    first_paren_new_line: no
  concurrent_401:
    disable: False
    align_left: True
    align_paren: False
```
With the following sample:
```
architecture rtl of my_entity is

begin

  wr_data <= (
    0,
    65535,
    32768
  );

end architecture rtl;
```
We obtain:
```
architecture rtl of my_entity is

begin

  wr_data <= (
  0,
  65535,
  32768
  );

end architecture rtl;
```

**Expected behavior**
The content of the array should be one indentation further than the parent indentation.

jeremiah-c-leary, 2022-12-21:
Hey @Scafir ,

I was looking at your configuration and if you change it to this:
```yaml
# sampleRules.yaml
rule:
  concurrent_012:
    first_paren_new_line: 'no'
  concurrent_401:
    disable: False
    align_left: 'yes'
    align_paren: 'no'
```
It should give you the correct indent.

--Jeremy

Scafir, 2023-01-05:
Hi @jeremiah-c-leary,

Following your instructions, it is working as intended, except for the fact that there is a collision with `concurrent_003`. This is most likely due to `first_paren_new_line: 'no'`.

Investigating a bit, this array syntax seems to cause issues even in the default case. As of vsg 3.14, running the sample without options falsely triggers `concurrent_011`.

Now that I am coming back from holidays with a fresh mind, I think that I was trying to split hairs a bit. I will use a more vsg-standard approach. As of my usecase, this ticket can be closed, but these rule collision may be worth looking into, especially for `concurrent_011`.

Regards,
Clyde

jeremiah-c-leary, 2023-04-01:
Hey @Scafir,

> Now that I am coming back from holidays with a fresh mind, I think that I was trying to split hairs a bit. I will use a more vsg-standard approach. As of my usecase, this ticket can be closed, but these rule collision may be worth looking into, especially for concurrent_011.

Would you be willing to create an issue to this behavior?

Thanks,

--Jeremy

closed 2023-04-19 by jeremiah-c-leary