jeremiah-c-leary/vhdl-style-guide #902: signal_400: crash with align_left = True and align_paren = False

Scafir, 2022-12-21:
**Environment**
 Fedora 37, VHDL Style Guide (VSG) version: 3.14.0+zip.file

**Describe the bug**
When `signal_400` is configured with align_left = True and align_paren = False, `vsg` crashes with:
`UnboundLocalError: cannot access local variable 'dExpectedIndent' where it is not associated with a value`.

**To Reproduce**
Rules:
```
rule:
  signal_400:
    align_left: True
    align_paren: False
```
sample:
```
architecture rtl of my_entity is

  signal sig_a : my_record (
    element1(7 downto 0),
    element2(3 downto 0)
  );

begin

end architecture rtl;
```
Run `vsg -c sampleRules.yaml -f sample.vhd` and get the error

**Expected behavior**
No crash

jeremiah-c-leary, 2022-12-21:
Hey @Scafir ,

Could you try changing `True` to `yes` and `False` to `no` in your configuration:
```yaml
rule:
  signal_400:
    align_left: 'yes'
    align_paren: 'no'
```
Thanks,

--Jeremy

Scafir, 2023-01-05:
Hi @jeremiah-c-leary,

Thank you for this syntax correction. Sorry for bothering you with this. 
It works as intended using the correction configuration values.

Regards,
Clyde

closed 2023-01-05 by Scafir