jeremiah-c-leary/vhdl-style-guide #911: Make `whitespace_013` silent on functional use of binary logic operators

staerz, 2023-01-04:
**Is your feature request related to a problem? Please describe.**
VSG (3.13.1) currently reports an error on the following VHDL-2008 functional use of binary logic operators:

```vhdl
...
architecture tb of tester is
  signal eof : std_logic_vector(1 downto 0);
...
begin
...
  check_loop : process
  begin
      while nand(eof) loop
  ...
```

**Describe the solution you'd like**
As of VHDL-2008, binary operators such as `nand`, `and` etc. can be used as a function and return the "nand"-ed, "and"-ed etc. result of the passed signal.

`whitespace_013` requires a space around the logical operators which usually isn't required for function calls.

The desired solution would be to make `whitespace_013` recognise the functional use and then not kick in.
(from the view of the implementation: if the expression within the parenthesis contains a space then it's a parenthesis for logical hierarchy and `whitespace_013` should be active, otherwise it's a functional use of the binary operator and `whitespace_013` should not kick in)

**Describe alternatives you've considered**
The alternative is to not care or to add local exceptions everywhere ...

**Additional context**
VHDL-2008 is great!

jeremiah-c-leary, 2023-11-12:
Good Evening @staerz ,

It has been awhile and I hope you are doing well.

I believe I found a case where I was not classifying the unary logical operator correctly in the following case:

```vhdl
  a<= b and and(some_signal);
```
So now the second `and` is properly classified and rule `whitespace_013` does not activate.

When you get a chance could you check this out on your end?

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-09:
Morning @staerz ,

Just wanted to ping you on this issue to see if you had time to check it out.

Thanks,

--Jeremy

jeremiah-c-leary, 2024-01-02:
Evening @staerz ,

Just wanted to ping you on this to see if you had time to check out the implementation.

Happy New Year,

--Jeremy

staerz, 2024-01-02:
Hey @jeremiah-c-leary,

I have some bad/sad news for you: I'm no longer dealing with VHDL as I left the project and the domain of firmware.
I will hence not be able to follow up on any of the issues that I have created.

You might ping @samdejong86 as he's still in the project and might take on this charge. If not, he might point you to another contact.

I thank you very much for all the effort that you put into VSG and wish you all best for its continuation as well as for you, personally!

Tschö, Steffen

jeremiah-c-leary, 2024-02-04:
Morning @staerz ,

> I have some bad/sad news for you: I'm no longer dealing with VHDL as I left the project and the domain of firmware.
I will hence not be able to follow up on any of the issues that I have created.

That is sad news for me.

I truly enjoyed our collaboration.  Your rule and documentation suggestions have made a lasting impression on the project.  I very much appreciate your contributions.

> You might ping @samdejong86 as he's still in the project and might take on this charge. If not, he might point you to another contact.

Thank you for the contact.  I will reach out.

> I thank you very much for all the effort that you put into VSG and wish you all best for its continuation as well as for you, personally!

Thank you for the kind words.  I wish you well in your next endeavor, and who knows, maybe our paths will cross again some day.

Best Regards,

--Jeremy

abyszuk, 2024-07-04:
Hi,
I'd like to report that this is still a problem in the newest VSG version.

jeremiah-c-leary, 2024-07-04:
Morning @abyszuk ,

By the newest version are you referring to 3.24.0 or the `issue-911` branch?

--Jeremy

jeremiah-c-leary, 2024-07-05:
Morning @abyszuk ,

I just merged trunk into the `issue-911` branch.  The branch includes updates to address this issue.  If you have time, could you check if the issue is resolved using the `issue-911` branch?

Regards,

--Jeremy

abyszuk, 2024-07-08:
Hello @jeremiah-c-leary 

I did some testing and while it works in most cases, it still fails in this one case:
```vhdl
  if or(adc_busy_int) = '1' or busy_src_i = BUSY_SRC_TIME_C then
```
where `adc_busy_int` is a vector.
To clarify - I've tested by cloning VSG repo, switching to that branch and running
`~/soft/vsg/bin/vsg -c my_config.yaml -f my_file.vhd`, but I also have a VSG version installed with `pip3`.
Should I uninstall it before running the test?

jeremiah-c-leary, 2024-07-09:
Evening @abyszuk ,

I tried your example in the following file:
```vhdl
  1
  2 architecture rtl of fifo is
  3
  4 begin
  5
  6   u_proc : process is
  7   begin
  8
  9     if (or(adc_busy_int) = '1' or busy_src_i = BUSY_SRC_TIME_C) then
 10     end if;
 11
 12   end process u_proc;
 13
 14 end architecture rtl;
```
and it passed without any errors.

Could you double check you are on the correct branch?

--Jeremy

abyszuk, 2024-07-09:
Hello @jeremiah-c-leary 

Yes, I'm on the correct branch and the code above passes the check.
However, if you remove external parenthesis it will fail:
```vhdl
if or(adc_busy_int) = '1' or busy_src_i = BUSY_SRC_TIME_C then
```
Sometimes it also fails when the `if` condition is nested inside `case when` statements, but I'm still trying to find reliable MWE.

jeremiah-c-leary, 2024-07-09:
Morning @abyszuk ,

It looks like my detection of unary operators needs to be updated.  I will need to think on it a bit.

--Jeremy

jeremiah-c-leary, 2024-07-10:
Evening @abyszuk ,

I pushed an update.  Please let me know if it fixes the issue you are seeing.

Thanks,

--Jeremy

abyszuk, 2024-07-10:
Hello @jeremiah-c-leary,

Yes, looks like it fixed all the problems I could find. Thanks again!

jeremiah-c-leary, 2024-07-11:
Evening @abyszuk ,

Thank you for validating this issue.  I will get it merged to master.

--Jeremy

PR #1046 (2024-07-11): Issue#911:  Improving detection of unary logical operators.
Resolves #911
closed 2024-07-11 by jeremiah-c-leary