jeremiah-c-leary/vhdl-style-guide #912: Subtypes with a resolution function crashes vsg

Scafir, 2023-01-26:
Hi @jeremiah-c-leary ,
I hope you had a nice start of year and that work will not be too harsh. I am making this bug report before I forget, but it is in no way urgent!

**Environment**

* HDL Style Guide (VSG) version: 3.14.0+zip.file
* Fedora Linux 37 (Workstation Edition)

**Describe the bug**
A subtype declaration with a resolution function (section 6.3 in the vhdl 2008 standard) : 
```vhd
subtype new_std_logic is (resolved) std_ulogic;
```
will crash vsg, as it expects a single word after the `is`.

**To Reproduce**
Steps to reproduce the behavior:
1. Use sample file:
```vhd
-- sample.vhd
library ieee;
use ieee.std_logic_1164.all;

package sample is
  subtype new_std_logic is (resolved) std_ulogic;
end package sample;
```
2. Run `vsg -f sample.vhd `
3. See error:
```
Error: Unexpected token detected while parsing subtype_declaration @ Line 5, Column 39 in file None
       Expecting : ;
       Found     : std_ulogic

[]
```

**Expected behavior**
The case when there is an extra word on the subtype declaration should pass without issues.

**Additional context**
I triggered this issue when running vsg against the [2008 standard libraries](https://opensource.ieee.org/vasg/Packages/-/tree/586ebeb9c3fcefd1ac9a07ce749d0e01e678503e/ieee). Maybe they could be added to the testcases?

Best Regards,
Clyde

jeremiah-c-leary, 2023-04-01:
Morning @Scafir,

> I hope you had a nice start of year and that work will not be too harsh. I am making this bug report before I forget, but it is in no way urgent!

Thanks for the kind words.  Work has been a little rough, spending way too many hours at it for the last three months.  I am hoping it will slow down a little in a month.

I pushed a fix for this to the `issue-912` branch.  Turns out the closing parenthesis was not being classified correctly.  Let me know how it works on your end.

> I triggered this issue when running vsg against the [2008 standard libraries](https://opensource.ieee.org/vasg/Packages/-/tree/586ebeb9c3fcefd1ac9a07ce749d0e01e678503e/ieee). Maybe they could be added to the testcases?

That is a great idea.  I will create a ticket so I do not forget.

Regards,

--Jeremy

jeremiah-c-leary, 2023-07-24:
Evening @Scafir ,

I am going to push this to master.

Thanks,

--Jeremy

PR #932 (2023-07-24): Issue#912:  Fixing resolution in subtype_declaration.
Resolves #912
closed 2023-07-24 by jeremiah-c-leary