jeremiah-c-leary/vhdl-style-guide #913: length_001 report seems to list "120" regardless of the config file

pdanders11, 2023-01-27:
**Environment**
State installed version of this project and your OS information.
3.14, 3.2.2
**Describe the bug**
A clear and concise description of what the bug is.

report says
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  length_001                | Warning    |        277 | Reduce line to less than 120 characters
  length_001                | Warning    |        278 | Reduce line to less than 120 characters
  length_001                | Warning    |        279 | Reduce line to less than 120 characters

but config says

  length_001:
    disable: false
    fixable: false
    indentSize: 2
    length: 170
    phase: 7

**To Reproduce**
Steps to reproduce the behavior:
update config to something besides 120

  length_001:
    disable: false
    fixable: false
    indentSize: 2
    length: 170
    phase: 7

**Expected behavior**
A clear and concise description of what you expected to happen.

report says
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  length_001                | Warning    |        277 | Reduce line to less than 170 characters
  length_001                | Warning    |        278 | Reduce line to less than 170 characters
  length_001                | Warning    |        279 | Reduce line to less than 170 characters

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

I think the check is correct but I second guess myself not see the config value in the report

jeremiah-c-leary, 2023-04-09:
Evening @pdanders11,

I pushed an update to this issue to the `issue-913` branch.  When you get a chance could you check it on your end and let me know if it is fixed.

Thanks,

--Jeremy

pdanders11, 2023-05-12:
Yes, this fixes the issue.  Thanks.

PR #936 (2023-05-13): Issue#913:  Fixing reporting of line length errors.
Resolves #913
closed 2023-05-12 by pdanders11