jeremiah-c-leary/vhdl-style-guide #916: Unexpected token detected while parsing subtype_declaration

olagrottvik, 2023-02-10:
**Environment**
`VHDL Style Guide (VSG) version: 3.14.0+zip.file`
`Linux 4.18.0-408.el8.x86_64`

**Describe the bug**

The following valid code

```
  subtype t_control_code is std_logic_vector(1 downto 0);
```
gives error:

```
Error: Unexpected token detected while parsing subtype_declaration
       Expecting : ;
       Found     : (
```

jeremiah-c-leary, 2023-04-01:
Morning @olagrottvik,

I tried your example in the following code snippet:
```vhdl
  1
  2 architecture rtl of fifo is
  3
  4   subtype t_control_code is std_logic_vector(1 downto 0);
  5
  6 begin
  7
  8 end;
```
and I did not receive the same error.  Could you try the above snippet on your side?

Thanks,

--Jeremy

jeremiah-c-leary, 2023-08-09:
Morning @olagrottvik ,

Just wanted to ping you on whether this is still an issue for you.

Thanks,

--Jeremy

olagrottvik, 2023-08-25:
Hi Jeremy,

Sorry for the extremely long delay. The issue is still there with the attached example file. Note, in a package file, not an architecture.

[test_pkg.vhd.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/12437002/test_pkg.vhd.txt)

```
VHDL Style Guide (VSG) version: 3.16.0+zip.file
```

Thanks for looking into it!

JHertz5, 2023-08-25:
Hi @olagrottvik. The file you've attached is not valid VHDL. It needs an ` is` after the package name. 

With your original file, I see the issue that you are describing. However, once I correct it, I no longer see the issue and VSG runs to completion.

olagrottvik, 2023-08-25:
Well, that's embarrassing... I don't know how that happened.
I no longer have the original example where this occurred for me the first time, so I guess without reproduction, you can ignore this issue. Sorry for wasting your time, and thanks for your efforts in developing this cool tool.

closed 2023-08-25 by olagrottvik