jeremiah-c-leary/vhdl-style-guide #917: Code tag "vsg_off" still leaves some rules active

MJoergen, 2023-02-15:
**Environment**
OS: Ubuntu 22.04.01 LTS
VSG version: 3.14.0+zip.file

**Describe the bug**
Despite having inserted the code tag "vsg_off" in the source file, when running VSG I still get errors related to the disabled section of code.

**To Reproduce**
Run VSG on the following file (with default configuration):
```
library ieee;
  use ieee.std_logic_1164.all;

entity tb is
end entity tb;

architecture simulation of tb is

  type t_rec is record
    val1 : integer;
    val2 : integer;
  end record t_rec;

  type t_rec_vector is array(integer range <>) of t_rec;

  -- vsg_off
  constant CONST : t_rec_vector :=
  (
    (val1 => 1, val2 => 2), -- Despite the tag vsg_off, VSG reports missing carriage return here
    (val1 => 3, val2 => 4), -- Despite the tag vsg_off, VSG reports missing carriage return here
    (val1 => 5, val2 => 6)  -- Despite the tag vsg_off, VSG reports missing carriage return here
  );
  -- vsg_on

begin

end architecture simulation;
```
When running `vsg -f issue.vhd` this gives the output:
```
================================================================================
File:  issue.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 521
Total Violations:    3
  Error   :     3
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  constant_016              | Error      |         19 | Add carriage return after open parenthesis.
  constant_016              | Error      |         20 | Add carriage return after open parenthesis.
  constant_016              | Error      |         21 | Add carriage return after open parenthesis.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

**Expected behavior**
I would expect VSG to completely ignore/skip the section of code between the tags "vsg_off" and "vsg_on", much like the well-known tags "synthesis_off" and "synthesis_on".

jeremiah-c-leary, 2023-04-09:
Morning @MJoergen,

I have pushed an update for this issue to the `issue-917` branch.  When you get a chance could you check it on your end and let me know if it is fixed on your end.

Thanks,

--Jeremy

MJoergen, 2023-04-13:
Verified, it works as expected.

PR #937 (2023-04-13): Issue#917:  Fixing missing code tags.
Resolves #917
closed 2023-04-13 by jeremiah-c-leary