jeremiah-c-leary/vhdl-style-guide #918: Add requirement for "end function" and "end procedure"

MJoergen, 2023-02-15:
The following code is legal VHDL, despite the missing keywords "procedure" and "function" after "end". Currently, VSG does not enforce the presence of these keywords:
```
library ieee;
  use ieee.std_logic_1164.all;

entity tb is
end entity tb;

architecture simulation of tb is

  procedure proc is
  begin

  end proc; -- Should require the keyword "procedure" after "end".

  function func return integer is
  begin

    return 0;

  end func; -- Should require the keyword "function" after "end".

begin

end simulation; -- Here vsg DOES report a missing keyword "architecture" after "end".
```
Running "vsg -f issue.vhd" gives the following result:
```
================================================================================
File:  issue1.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 153
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  architecture_010          | Error      |         23 | Add *architecture* keyword.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

I currently see no configuration option to make VSG enforce the presence of the keywords "procedure" or "function" after "end". This would be really nice to have, so as to increase readability, in much the same way that it already does require the keyword "architecture".

**Describe the solution you'd like**
I would like two separate rules: `function_***` and `procedure_***` (similar to `architecture_010`) where the user can optionally require the presence of the corresponding keyword after end.

jeremiah-c-leary, 2023-04-09:
Evening @MJoergen,

I pushed an update to the `issue-918` branch.  When you get a chance could you check it out on your end and let me know if it works for you?

Thanks,

--Jeremy

MJoergen, 2023-04-13:
Using the default settings, I now get the following output.

```
================================================================================
File:  issue.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 155
Total Violations:    3
  Error   :     3
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  procedure_012             | Error      |         12 | Add procedure keyword
  function_018              | Error      |         19 | Add function keyword
  architecture_010          | Error      |         23 | Add *architecture* keyword.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
I note a slight difference in that "architecture" is surrounded by asterisks, but "procedure" and "function" are not.

Furthermore, if I now apply the "--fix" option, I still get the following output:
```
================================================================================
File:  issue.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 155
Total Violations:    2
  Error   :     2
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  procedure_012             | Error      |         12 | Add procedure keyword
  function_018              | Error      |         19 | Add function keyword
----------------------------+------------+------------+--------------------------------------
```
So, apparently, there is no "auto-fix" for the missing "procedure" and "function" keywords.

In conclusion, I would like the output to be consistent, and I would like auto-fix enabled for these additional keywords.

jeremiah-c-leary, 2023-04-13:
Morning @MJoergen,

I will double check my tests as I thought I had this covered.

--Jeremy

jeremiah-c-leary, 2023-04-15:
Morning @MJoergen,

I believe I found the issue and pushed an update to the branch.  Could you check it out and let me know if it fixes the issue on your side?

Thanks,

--Jeremy

MJoergen, 2023-04-18:
Works as expected now. Thank you!

PR #939 (2023-04-19): Issue#918:  Adding rules for function and procedure optional keywords.
Resolves #918
closed 2023-04-19 by jeremiah-c-leary