jeremiah-c-leary/vhdl-style-guide #919: constant_012 incorrectly wants to indent empty line

MJoergen, 2023-02-15:
**Environment**
OS: Ubuntu 22.04.01 LTS
VSG version: 3.14.0+zip.file

**Describe the bug**
VSG wants me to indent an empty line, i.e. insert spaces into a line that is otherwise empty. This is silly and leads to trailing white space.

**To Reproduce**
Run `vsg -f issue.vhd` on the following file:
```
library ieee;
  use ieee.std_logic_1164.all;

entity dut is
end entity dut;

architecture simulation of dut is

  type t_rec is record
    val1 : integer;
    val2 : integer;
  end record t_rec;

  type t_rec_vector is array(integer range <>) of t_rec;

  constant CONST : t_rec_vector :=
  (
    (
      val1 => 1,
      val2 => 2
    ), -- line 21

    ( -- line 23
      val1 => 3,
      val2 => 4
    )
  );

begin

end architecture simulation;
```

The results obtained are:
```
================================================================================
File:  issue.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 521
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  constant_012              | Error      |         22 | Indent with 4 space(s)
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

**Expected behavior**
VSG (and in particular the rule constant_012) should recognize the empty line and not try to inserts spaces.

jeremiah-c-leary, 2023-04-09:
Evening @MJoergen,

I pushed an update for this to the `issue-919` branch.  When you get a chance could you check it out and let me know if it works on your end?

Thanks,

--Jeremy

MJoergen, 2023-04-13:
Verified, works as expected.

jeremiah-c-leary, 2023-04-13:
Awesome,

I will get this merged to master.

--Jeremy

closed 2023-04-13 by jeremiah-c-leary