jeremiah-c-leary/vhdl-style-guide #921: [QUESTION - FUNCTION]: Function call parameter alignment

sibeov, 2023-02-23:
# What is your question?
Hi.

Tried to find relevant rule, but to my (probably ignorance) avail I could not. I'm trying to disable the automatic function call parameter 
alignments.

## Example
```vhdl
-- Before VSG alignment
f_some_function(
  some_param       => some_value,
  some_other_param => some_other_value
);

-- After VSG Alignment
f_some_function(
                          some_param       => some_value,
                          some_other_param => some_other_value
               );
```

Which rule(s) do I disable to dissable this behaviour?

Kind Regards
sibeov

jeremiah-c-leary, 2023-04-09:
Greetings @sibeov,

I could see why finding the correct rule would be difficult.  At the moment VSG can not tell the difference between a function call and a procedure call and classifies all functions as procedure calls.

The rule `procedure_call_400` is performing the alignment above.  If you want to keep the first formatting you can use the following configuration:

```yaml
rule:
  procedure_call_400:
    align_left : 'yes'
    align_paren : 'no'
```

Regards,

--Jeremy

sibeov, 2023-04-13:
Thanks @jeremiah-c-leary ! This works!

Is there any roadmap to split them into function call and procedure call? If so I could open an issue?

jeremiah-c-leary, 2023-04-13:
Morning @sibeov ,

That is something I have though long and hard about.  In order to tell the difference between procedure and function calls would require having knowledge of the available procedure and function specifications.  These could be defined in a package in a separate file.  Currently VSG processes a single file at a time, so this limits it's ability to properly classify procedure and function calls.

My main concern for changing the current behavior is more with usage.  All dependent files would need to be included and processed in order to analyze a single file.  This could be handled on the command line, but then how would it work in an editor?

So at the moment there are no roadmap plans to differentiate between function and procedure calls.

Regards,

--Jeremy

jeremiah-c-leary, 2023-11-04:
Morning @sibeov ,

I am revisiting older issues and was wondering if you ware okay with closing this one.

I will plan on closing this after Nov 18th if I do not hear back from you.

Thanks,

--Jeremy

sibeov, 2023-11-05:
All good! This works for me now. Thanks for asking. 😊👌

closed 2023-11-05 by jeremiah-c-leary