jeremiah-c-leary/vhdl-style-guide #922: Negative time values catched by whitespace_011

erick166, 2023-02-24:
**Environment**
VHDL Style Guide (VSG) version: 3.13.1

**Describe the bug**
Negative time values like `-1 ns` are reported by this rule.

**To Reproduce**
Steps to reproduce the behavior:
```
p_test : process is
  variable v_timestamp : time := -1 ns;
begin

  if v_timestamp /= -1 ns then
    -- do something
  end if;

end process p_test;
```

**Expected behavior**
I would expect -1 ns to be a valid assignment, and not - 1 ns.

jeremiah-c-leary, 2023-04-09:
Morning @erick166,

I pushed an update to this issue to the `issue-922` branch.  When you get a chance could you check it out on your end and let me know if it works on your end?

Thanks,

--Jeremy

erick166, 2023-04-13:
Hi,

The negative value in the **if** statement is okay now, but negative values in the declaration part are still reported. E.g.

```
constant C_MIN_RANGE : integer := -2 ** 8;
variable v_timestamp : time := -1 ns;
```

jeremiah-c-leary, 2023-04-15:
Morning @erick166,

I pushed another fix for this to the branch.  When you get a chance could you check it out and let me know how it works for you?

Thanks,

--Jeremy

erick166, 2023-04-17:
Hi,

It works now with the new fix, thanks!

PR #938 (2023-04-17): Issue_922:  Fixing mis-classification of signs as adding operators.
Resolve #922
closed 2023-04-17 by jeremiah-c-leary