jeremiah-c-leary/vhdl-style-guide #923: Indentation for protected types

erick166, 2023-02-24:
**Environment**
VHDL Style Guide (VSG) version: 3.13.1

**Describe the bug**
Indentation for protected type structures is not respected.

**To Reproduce**
```
type t_protected_fifo is protected

  procedure reset_fifo (
    void : t_void
  );

end protected t_protected_fifo;

type t_protected_fifo is protected body

  variable dummy : std_logic := '0';

  procedure reset_fifo (
    void : t_void
  ) is
  begin
    -- do something
  end procedure reset_fifo;

end protected body t_protected_fifo;
```

**Expected behavior**
I think it should be possible to indent the code inside the protected type.

jeremiah-c-leary, 2023-04-08:
Evening @erick166,

I pushed an update for this issue to the `issue-923` branch.  When you get a chance could you check it out on your end and let me know if it works.

Thanks,

--Jeremy

erick166, 2023-04-13:
Hi,

I just tested the fix and it works :)

jeremiah-c-leary, 2023-04-13:
Awesome,

I will get this merged to master.

--Jeremy

PR #935 (2023-04-13): Issue#923:  Adding indenting of protected type declarations.
Resolved #923