jeremiah-c-leary/vhdl-style-guide #924: Support for CamelCase

JimLewis, 2023-02-26:
Just a quick review seems to indicate that it does not support any form of UpperCamelCase or lowerCamelCase.

Am I correct in my assessment?   Any plans to support these?

I have 30 years of using VHDL with UpperCamelCase and augmenting it with special leading characters such as 'n' for active low.   In addition '_' is not in a so convenient place for US keyboards.   If US keyboards were like European keyboards and '_' was next to '.' then I probably would be using '_'.   However, even worse - when I started the SUN workstation keyboards had the '_' on the numeric keypad.

jeremiah-c-leary, 2023-04-09:
Hey @JimLewis,

> Am I correct in my assessment?

You are correct, the only casing supported is either upper or lower case.  There are a couple of consistent case rules that may be helpful.

> Any plans to support these?

What kind of support were you looking for?

I have concerns that I could reliably turn a random string into camel case without some kind of dictionary. 

Regards,

--Jeremy

JimLewis, 2023-04-09:
Consistent case is good.   No errors on usage of either lowerCamelCase or UpperCamelCase.   Flag if names are not consistent, but it may require user interaction to fix.

jeremiah-c-leary, 2023-04-10:
Morning @JimLewis,

I can support checking a string follows camel case, but I can not tell if the camel case is correct.  For example `fifoWriteEnable` and `fiFowRiteenAble` would both be valid lowerCamelCase.

Would that be a sufficient implementation?

--Jeremy

t-aras, 2023-05-31:
Hei @jeremiah-c-leary 

I would also be interested in that.
Just checking, that it is either a _camelCase_ or a _PascalCase_ would help me alot.

If possible, it would also be beneficial if we I could configure to check that it is not _kebab-case_ or _snake_case_.

jeremiah-c-leary, 2023-10-28:
Good Afternoon @JimLewis and @t-aras ,

I pushed an update for this issue to the `issue-924` branch.  I added two options `camelCase` and `PascalCase` to the `case` option.  If the token does not match the case then an error will be issued.

I need to ensure the rest of the case rules are working, but could you give this a try and let me know what you think?

Thanks,

--Jeremy

t-aras, 2023-10-30:
Hey @jeremiah-c-leary 

Thank you for the work you've put into this. I just gave it a try, and it looks good from my side. However, we won't be able to use it as it currently stands. The case rule appears to evaluate the suffix, which consistently results in errors. Our suffix begins with a `_`, followed by a combination of uppercase letters.

I believe I'll need to implement a special `case` option tailored for our use case. This option would either remove the suffix or match `camelCase` and `PascalCase` while accommodating our specific suffix pattern following the name.

Do you by any chance have a solution or suggestion for this issue? If not, I'll work on it myself.

Best
t-aras

Hey @jeremiah-c-leary,

Thank you for the work you've put into this. I just gave it a try, and it looks good from my side. However, we won't be able to use it as it currently stands. The case rule appears to evaluate the suffix, which consistently results in errors. Our suffix begins with a `_`, followed by a combination of uppercase letters.

I believe I'll need to implement a special case option tailored for our use case. This option would either remove the suffix or match camelCase and PascalCase while accommodating our specific suffix pattern following the name.

Do you by any chance have a solution or suggestion for this issue? If not, I'll work on it myself.

jeremiah-c-leary, 2023-10-31:
Morning @t-aras ,

Question on your first issue regarding the prefix.  So my understanding is you would like to check for PascalCase but with a prefix.  Something like this:
```vhdl
  constant c_AddressWidth : natural := 32;
```
I can update the case rule to allow for prefixes.

On your second test regarding signal names like `FFTSymbInfo`, I could not find anything description of camelCase or PascalCase that limited the number of capital letters to 1.  So I will update the regex as you propose.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-10-31:
Hey @t-aras ,

How should we handle a name like:  `SymbInfoFFT`?  Would that still pass a Pascal Case check?

--Jeremy

t-aras, 2023-10-31:
Good Morning @jeremiah-c-leary 

We are using suffixes not prefix. Like for example Rst_RI or Clk_CI.
I have defined all allowed suffixes in the suffix rules. For example `port_025` or `signal_600`.
So like this:
```vhdl 
  signal Clk_C      : std_logic;
```
And for ports:
```vhdl
entity my_module is
  port (
    Clk_CI     : in  std_logic;  -- Clock input
    Rst_RBI    : in  std_logic;  -- Reset input
  );
end entity my_module;
```

Concerning `SymbInfoFFT`, the regular expression I proposed does accept this scenario. 
It also falls short in supporting single-letter identifiers. Your initial regex does support single letter names.
Mine will also not support all uppercase naming. But I guess according to the naming you provided, it anyway shouldn't accept all uppercase names. I'll have to see if I can come up with an expression that rejects names ending with a capital letter and accepts single letter names.

jeremiah-c-leary, 2023-11-01:
Morning @JimLewis and @t-aras ,

I pushed an update where I added support for prefixes, suffixes and case exceptions.   Let me know how it works on your end.

I am going to be spending some time refactoring the code to reduce duplication and make it more efficient.

Regards,

--Jeremy

t-aras, 2023-11-09:
Hello @JimLewis,

Apologies for the delayed response; I only recently had the opportunity to run tests.

The current implementation generally meets our needs well. However, the handling of suffixes presents a significant challenge for our use case. We employ a variety of letter combinations in our suffixes, which would typically be a prime candidate for regex matching. As it stands, I would have to enumerate all permissible letter combinations in the port_025 rule, and then replicate this for each case rule — a rather cumbersome process. Should I perhaps open a new issue to explore this further? What are your thoughts?

On a positive note, the checks for PascalCase and camelCase are functioning correctly, which is much appreciated!

Thank you for your efforts!

jeremiah-c-leary, 2023-11-11:
Morning @t-aras ,

> On a positive note, the checks for PascalCase and camelCase are functioning correctly, which is much appreciated!

That is good news.

> I would have to enumerate all permissible letter combinations in the port_025 rule, and then replicate this for each case rule — a rather cumbersome process. Should I perhaps open a new issue to explore this further? What are your thoughts?

Could you create a new issue for a regex on suffixes and prefixes.  That would allow me to close this issue and then we can discus implementation options for the regex in the new issue.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-11-14:
Good Morning @JimLewis and @t-aras ,

There is another issue which will require an update to the case logic, so I am going to merge this to master to reduce merge conflicts.  If there are any additional update required for cameCase or PascalCase, please create another issue and I will resolve it.

Thanks,

--Jeremy

PR #1037 (2023-11-14): Issue#924:  Adding ability to check for camelCase and PascalCase.
Resolves #924