jeremiah-c-leary/vhdl-style-guide #926: Element association inside constant declaration

erick166, 2023-03-02:
**Environment**
VHDL Style Guide (VSG) version: 3.14.0

**Describe the bug**
The rule **element_association_100** checks for spaces between the `others` keyword and the `=>` assignment inside a multi-line constant declaration.

**To Reproduce**
```
constant C_ADDRESS_CONTROL : natural := 0;
constant C_ADDRESS_DATA    : natural := 1;
type t_address_en is array (0 to 31) of boolean;

constant C_DEFAULT_VALUES : t_address_en := (
  C_ADDRESS_CONTROL => false,
  C_ADDRESS_DATA    => true,
  others            => false
);
```

**Expected behavior**
The `=>` assignment should have the same alignment as the other elements inside the constant declaration.

jeremiah-c-leary, 2023-10-29:
Good Morning @erick166 ,

You can control the acceptable whitespace using the following configuration:
```yaml
  rule:
    element_association_100:
      number_of_spaces: '>=1'
```
This will allow more than more space after the `others` keyword.

I am working on an alignment rule for constant declarations as part of issue #999 

Let me know if the above configuration works for you or if there is more work to be done.

Thanks,

--Jeremy

erick166, 2023-11-14:
Hi,

Yes, that configuration works, thanks!

closed 2023-11-15 by jeremiah-c-leary