jeremiah-c-leary/vhdl-style-guide #927: Add a comment rule that check for the presence of FIXME or TODO in code comments

BrunoJJE, 2023-03-07:
Would be cool to be warned in the case we have left TODO or FIXME in code comments.

Could be a configurable list of words that is set by default to FIXME and TODO.

jeremiah-c-leary, 2023-04-01:
Morning @BrunoJJE,

That sounds like a good idea.  I am thinking the rule would be a Warning instead of an Error.
I will add this to the backlog.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-04-08:
Evening @BrunoJJE,

I pushed an update for this issue to the `issue-927` branch.  When you get a chance could you check it out.  I added a new rule `comment_012` which is disabled by default so you will need to have a configuration to enable it.

Thanks,

--Jeremy

BrunoJJE, 2023-04-11:
Tested with the issue-927 branch and a configuration file that set disable to false for this rule.
Working fine with the default FIXME and TODO keywords.
Working fine with custom keywords defined in the configuration file for this rule.
Thanx.

PR #934 (2023-04-11): Issue#927:  Adding checks for keywords in comments.
Resolves #927
closed 2023-04-11 by jeremiah-c-leary